KiCad/kicad-footprints #396: 5.0.0-rc1 repository unusable on Linux

calebns, 2018-02-26:
After cloning the fresh repo, the git sees that the file WDFN-8-2x2mm is not staged for commit.
[image]

It seems that it contains invalid line endings which makes the Linux git to interpret the file as being changed. What suprises me, the git-2.7.4 gets crazy at this point and refuses to either check out anything or to unstage/revert changes to that file.
The following patch fixes the issue by removing 0xD character from line endings, i.e. makes the line endings compatible with the one found in other files.

[0001-test.log](https://github.com/KiCad/kicad-footprints/files/1760029/0001-test.log)

jkriege2, 2018-02-26:
could you please open  a PR for the fix you proposed?

Thanks,
JAN

poeschlr, 2018-02-27:
Fix merged

closed 2018-02-27 by poeschlr