KiCad/kicad-footprints #412: Where should stacking stripe footprints live?

evanshultz, 2018-03-06:
@poeschlr @jkriege2 @Ratfink 
I've written a small script to product stacking stripes as described on pages 3-5 of https://www.speedingedge.com/PDF-Files/Test%20Structures.pdf.

Here is an image:
[image]

What should the library name be for these? I would think `Test_` would be a good prefix. We could have `Test_Points.pretty`, `Test_Stripes.pretty`, etc. That would mean renaming one existing repo.

Any objections? Or suggestions? I'm a bit unsure of the best name of the footprints, so suggestions there are welcome too!

Here is the current script:
```
import sys
import os
import argparse

sys.path.append(os.path.join(sys.path[0], "../.."))  # enable package import from parent directory

from KicadModTree import *  # NOQA

def round_to(n, precision):
    correction = 0.5 if n >= 0 else -0.5
    return int( n/precision+correction ) * precision

if __name__ == '__main__':

    # handle arguments
    parser = argparse.ArgumentParser()
    parser.add_argument('layers_max', help='the maximum layer count to generate strips (default=32)', type=int, nargs='*', default=32)
    parser.add_argument('inner_layer_prefix', help='prefix for inner layers (default=In)', type=str, nargs='*', default="In")
    parser.add_argument('-v', '--verbose', help='show extra information while generating the footprint', action='store_true')
    args = parser.parse_args()
    
    for layer_count in range(2, args.layers_max + 2, 2):
        
        footprint_name = 'Test_Layer_Strips_{lc}layer'.format(lc=layer_count)
        
        print('Building {:s} '.format(footprint_name))
        
        # calculate working values
        mm_inch = 25.4
        
        trace_width = 0.005 * mm_inch
        trace_depth = 0.05 * mm_inch
        stripe_width = trace_depth
        stripe_x_offset = trace_width / 2.0 + trace_depth
        
        courtyard_x = stripe_x_offset + stripe_width * layer_count + 1 + 0.025
        courtyard_y = trace_depth / 2.0 + 0.025
        
        # initialise footprint
        kicad_mod = Footprint(footprint_name)
        kicad_mod.setDescription('Test structure, layer strip, {:g} layers (https://www.speedingedge.com/PDF-Files/Test%20Structures.pdf)'.format(layer_count))
        kicad_mod.setTags('test structure layer strip'.format(layer_count))

        # set general values
        kicad_mod.append(Text(type='reference', text='REF**', size=[1,1], at=[0, -1], layer='F.SilkS'))
        kicad_mod.append(Text(type='user', text='%R', size=[1,1], at=[0, 0], layer='F.Fab'))
        kicad_mod.append(Text(type='value', text=footprint_name, at=[0, 1], layer='F.Fab'))

        # place guide line for board edge
        kicad_mod.append(Line(start=[-1, 0], end=[stripe_width * layer_count + 2, 0], layer='Dwgs.User', width=0.15))
        
        # F.Cu elements
        kicad_mod.append(Line(start=[0, -trace_depth / 2.0], end=[0, trace_depth / 2.0], layer='F.Cu', width=trace_width))
        kicad_mod.append(FilledRect(start=[stripe_x_offset, -trace_depth / 2.0], end=[stripe_x_offset + stripe_width, trace_depth / 2.0], layer='F.Cu'))
        
        # inner layer elements
        for layer in range(1, layer_count - 1):
            kicad_mod.append(Line(start=[0, -trace_depth / 2.0], end=[0, trace_depth / 2.0], layer=args.inner_layer_prefix + str(layer) + '.Cu', width=trace_width))
            kicad_mod.append(FilledRect(start=[stripe_x_offset, -trace_depth / 2.0], end=[stripe_x_offset + stripe_width * layer, trace_depth / 2.0], layer=args.inner_layer_prefix + str(layer) + '.Cu'))
            
        # B.Cu elements
        kicad_mod.append(Line(start=[0, -trace_depth / 2.0], end=[0, trace_depth / 2.0], layer='B.Cu', width=trace_width))
        kicad_mod.append(FilledRect(start=[stripe_x_offset, -trace_depth / 2.0], end=[stripe_x_offset + stripe_width * layer_count, trace_depth / 2.0], layer='B.Cu'))
            
        # create courtyard
        kicad_mod.append(RectLine(start=[-trace_width / 2.0 + 0.025, -courtyard_y], end=[courtyard_x, courtyard_y], layer='F.CrtYd', width=0.05))

        # add model
        kicad_mod.append(Model(filename="${{KISYS3DMOD}}/Test_Structures.3dshapes/{:s}.wrl".format(footprint_name), at=[0, 0, 0], scale=[1, 1, 1], rotate=[0, 0, 0]))

        # print render tree
        if args.verbose:
            print(kicad_mod.getRenderTree())

        # write file
        file_handler = KicadFileHandler(kicad_mod)
        file_handler.writeFile('{:s}.kicad_mod'.format(footprint_name))

```

poeschlr, 2018-03-06:
What am i seeing here? You do not show the layer tap so i can't tell what the colors mean. Are these small squares on all copper layers?

Ratfink, 2018-03-06:
I think @evanshultz was suggesting we rename `TestPoint.pretty` to `Test_Point.pretty`, along with creating this new `Test_Stripe.pretty` (names changed to singular as per G1.5).  I think that's a reasonable proposal, even though they're for different sorts of testing (manufacturing tests vs. circuit testing).

evanshultz, 2018-03-06:
@poeschlr 
Hmm.... See the names I suggested above and refined by Clayton to singular. But seeing that Oliver changed `Test_Point` to `TestPoint` maybe this one should go into `TestStripe.pretty` and no repos are renamed.

I'm suggesting to use a `Test` prefix for repos that house footprints used for testing. If `Test` is at the end they aren't grouped together alphabetically.

And for the footprint names? `TestStripe_1.27x1.27mm_??Layer`?

Ratfink, 2018-03-06:
I'd suggest adding the test trace's width to the footprint name somewhere.  Maybe something like `TestStripe_1.27x1.27mm_Trace0.127mm_??Layer`.

evanshultz, 2018-03-14:
@poeschlr @Ratfink @jkriege2 
Any other thoughts? If we can settle on where to upload and what to name them, I can be done with this.

Ratfink, 2018-03-14:
Underscore or no underscore in the library name, it doesn't make much difference to me.  I could argue for either way making more sense.  No underscore does have the advantage of not having to rename anything to maintain consistency, so I guess I'd favor that option slightly.

I've got nothing else to add to my earlier naming suggestion.