KiCad/kicad-footprints #425: PLCC footprint issues

evanshultz, 2018-03-12:
On all PLCC32 packages I see:
- Pin 1 mark is misaligned (some too far left and some too far right).
- Silk offset from fab lines is not consistent.
- The bevel on F.SilkS doesn't have the same offset from F.Fab as orthogonal lines.
- Some silk is overly close to pads.
- Some silk is overly far from pads.
- The ref des and value aren't outside the courtyard, which is typical.
- No datasheet or other info in the description.
- Socketed versions don't mention they are socketed in the description.
- SMD socket has silk under the socket body.

[image]

I don't see a script at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Packages, but it's probably best to script them.

evanshultz, 2018-04-29:
@poeschlr 
https://github.com/KiCad/kicad-footprints/pull/515 added a few scripted PLCC footprints. Are you planning to add others? Are any or all of the issues above being addressed?

poeschlr, 2018-04-29:
The points above are addressed (as far as they applied) for the 3 footprints i added so far. As soon as my PR is merged anyone should be able to add further footprints to the generator. (I don't think i will have that much time in the next few weeks as the large "find a fitting" footprint issue is not yet fully resolved and i should also work on uni stuff.)

As the generator does derive the footprints via IPC standards directly from the package dimension, a dimensioned drawing of the package is enough, no need to search for a suggested footprint.