KiCad/kicad-footprints #426: What is the recommended folder for footprints and symbols?

arildj78, 2018-03-12:
I'm new to KiCad and have downloaded a nightly build to be able to do circuit simulation. One problem when entering this project is that I can't find the documentation on where to put symbols, footprints and models.

It is suggested to clone the different projects to get the latest change, but could you include some folder names for Win10 in the Read.me?

poeschlr, 2018-03-12:
There is no requirement. You can point kicad to look for libs in any location you like.

for footprints i wrote a description over at the forum https://forum.kicad.info/t/how-can-i-install-a-specific-version-of-the-footprint-library/9269

For symbols it is similar but instead of copying the fp-lib-table file you need to copy the sym-lib-table file.
Or use the symbol library wizard/manager found in eeschema->(setup or preferences menu. I don't know where they put it in the current nightly version.)

Ratfink, 2018-05-04:
Closing this, as it's been inactive for nearly two months, and as Rene pointed out, there is no specific recommended path for the library.

closed 2018-05-04 by Ratfink