KiCad/kicad-footprints #436: How to name footprints missing a single pin?

evanshultz, 2018-03-17:
@poeschlr @jkriege2 @Ratfink @Misca1234 
Started from a conversation at https://github.com/KiCad/kicad-symbols/pull/322.

Some parts are missing a pin. In some cases the footprint for the part will be manufacturer-specific. But in other cases it's not, or it doesn't make sense to do that because it would result in multiple footprints which are otherwise identical.

Let me take the common DIP8 package as an example and show some datasheets with a missing pin:
- http://www.onsemi.com/pub/Collateral/NCP1076A-D.PDF: either pin 3 or pin 6
- https://ac-dc.power.com/system/files_force/product-docs/tinyswitch-iii_family_datasheet.pdf: pin 3
- https://www.mouser.com/ds/2/328/lnk520-5820.pdf: pin 6
- http://www.st.com/content/ccc/resource/technical/document/datasheet/cd/d4/93/15/03/e1/40/81/CD00279018.pdf/files/CD00279018.pdf/_jcr_content/translations/en.CD00279018.pdf: pin 6
- https://www.infineon.com/dgdl/Infineon-ICE3AR0680VJZ-DS-v02_01-en.pdf?fileId=db3a304341e0aed00141ffe31ad53057 pin 6
- http://www.onsemi.com/pub/Collateral/NCP1216-D.PDF: pin 7
- http://rohmfs.rohm.com/en/products/databook/datasheet/ic/power/isolated_converter/bm2pxx2-e.pdf pin 6
- http://www.onsemi.com/pub/Collateral/NCP1337-D.PDF: pin 7
- https://www.diodes.com/assets/Datasheets/AP3983BCD.pdf: pin 7
- https://www.dialog-semiconductor.com/sites/default/files/iw1818_product_summary.pdf: pin 6
- https://www.diodes.com/assets/Datasheets/AP3983BCD.pdf: pin 7

We would need at least three, and possibly up to six, footprints for the parts above. Three with pin numbers up to 8 but missing pin 3, or 6, or 7, and another three with pin numbers up to 7 and missing the same three pins.

For packages with hidden or deleted pads according to IPC 7351, footprint names would be `DIP-8-7_W7.62mm` or `DIP-7-8_W7.62mm`. We would then need a way to capture the missing pins as there is no way to differentiate which pin is missing.

I see two ways this could be done, using one example:
1.  `DIP-8-7_W7.62mm_N7` where `N7` is the nomenclature for pin 7 being removed
2.  `DIP-8-7-N7_W7.62mm`

Does anyone have a preference? I think they're both equally easy to parse in a footprint filter. I like option 2 better because the pin information stays together, but I could be swayed.

Also, I notice at https://github.com/KiCad/kicad-footprints/tree/master/Package_DIP.pretty there are DIP6 footprints with pin 5 missing. Once this settles, they may not conform and may need to be renamed (and perhaps fixes to the symbol using those footprints).

Ratfink, 2018-03-17:
Option 2 looks like the most reasonable choice to me.  I see a bit of inconsistency with current KLC naming though, I'd think it should be `DIP-7-8-N7-W7.62mm` to minimize the change.

evanshultz, 2018-03-17:
See the sections for hidden pins and deleted pins in IPC 7351 Land Pattern Naming Conventions, as mentioned above. I don't know why Oliver only put one of them in KLC but it's a omission. This is the nomenclature that will account for having the highest pin number be 7 or 8.

Though if the highest pin number is 7 there isn't a pin that can be removed.

Now that I read IPC7351 carefully again it's actually saying `DIP-8_7N_xxxx`, with an underscore and a `N`. It still doesn't specify which pin is missing, and adding `N7` or `N6` in there looks even more ugly. Of course we don't have to strictly follow IPC.

evanshultz, 2018-03-17:
https://github.com/KiCad/kicad-footprints/pull/405 is a footprint where we used a subset of the pin numbers to make it work. In this case we went manufacturer-specific, but it's just another example.

poeschlr, 2018-03-17:
Option 3 looks good enough for me. (I am sure there will be some strange case that we can not represent with it but we can look into that once we are that far.)

Option 2 would also work but it seems to have redundant information and it looks quite strange. (But it would be more compatible with what we currently have.)

Ratfink, 2018-03-17:
You're right, option 3 does contain all the necessary information, and is the most concise.  It may require changing some footprint filters though.  As long as we have a way to find all the symbols that would need updating for the new naming convention, I'd be happy with option 3.

evanshultz, 2018-03-19:
OK. I'm fine with that for a symbol/footprint that uses up to pin number 8 in the example above. I think IPC would actually want to name it `DIP-7-8N_W7.62mm`, but that doesn't catch the exact pin that's missing. Option 3 is more concise and intelligent.

But what do we do for symbols/footprints that skip the missing pin so the highest pin number used in the datasheet is 7? Perhaps we force them to use the convention above, as I believe there's only a few and can be handled on a case-by-case basis? Any better ideas?

evanshultz, 2018-05-31:
See http://www.analog.com/media/en/technical-documentation/data-sheets/3638fa.pdf submitted at https://github.com/KiCad/kicad-symbols/pull/637. There's an example where multiple pins are missing and we didn't settle on a scheme above. Any thoughts?

I don't find the package drawing by itself online (I could have missed it) but the full-pin version is at http://www.analog.com/media/en/package-pcb-resources/package/pkg_pdf/ltc-legacy-msop/05081667_F_MSE16.pdf.

Perhaps in that case we go manufacturer-specific can use `Analog_MSE16-12` or something (since ADI bought LT, or else stick with `Linear_*`)?

Misca1234, 2018-06-10:
By making these stray foot prints manufacture specific (as stated above) there will be no problem by the naming convention.

Another way would be to keep the naming convention and just prefix or suffix to indicate something is different with it.
Like this
DIP_VX-8-7-W7.62mm
DIP-8-7-W7.62mm_VX

Where X can be a leap number for each of the foot prints that have various missing pins