KiCad/kicad-footprints #439: KLC: Silkscreen-to-pad clearance?

evanshultz, 2018-03-19:
https://github.com/KiCad/kicad-footprints/pull/354 fixed a footprint by pushing silk lines farther from the pads.

[KLC F5.1.2](http://kicad-pcb.org/libraries/klc/F5.1/) requires at least 0.2mm clearance. IPC 7351C is [recommending the silk line width](http://www.ocipcdc.org/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf) clearance from pads, which would be a minimum of 0.12mm (KLC allows some deviation but I believe almost all footprints are 0.12mm silk line width).

Should KLC be updated? Move the minimum from 0.2mm to 0.12mm in accordance with IPC 7351C?

Ratfink, 2018-03-19:
The main thing that PR was fixing was the reversed X and Y dimensions in the footprint name.  But you're right, it also moved the silkscreen lines, as they were very close to the pads.  They would have been too close by the rule you're suggesting as well; the current rule just made them need a bit extra clearance.

Here's the discussion from the old kicad-library repository that resulted in this KLC rule in the first place: https://github.com/KiCad/kicad-library/issues/1422

I notice that you mentioned the same thing in the old discussion also, but that was mostly ignored after 0.2 mm was suggested.  I'm certainly not opposed to revising the rule to allow the reduced clearance recommended by IPC, just so long as it's done in a backwards-compatible way.

poeschlr, 2018-03-22:
The current silk-pad clearance did not come from IPC recommendations. We choose 0.2mm as we could then use a "feature" of kicad 4.x to easily check it. (Kicad 4.x and at least older nightlies shows the mask clearance in the opengl canvas of footprint editor as 0.2mm by default.) And because this is compatible with the fab i use.

If we want to update the KLC i would suggest to (for now) add either a range or document that old footprints used 0.2mm.
If we go with the allowed range we could specify some relation between part size and clearance. As for larger parts i think it is possible to have a larger clearance while still having well defined silk without resulting in a much larger covered area.