KiCad/kicad-footprints #444: Footprint not linked with last packages3D version

hildogjr, 2018-03-20:
Recently the packages3D folder has been restructured. The most part of the footprint not load the 3D visualization because of error in the '3D Shape Name' file.

poeschlr, 2018-03-20:
Could you give at least one example?

As far as i know most 3d models in the [packages3d repo](https://github.com/KiCad/kicad-packages3D) should be correct. The old kicad-library repo is not indented to work with the new footprint organization as it is used to keep the kicad 4 legacy state unchanged.

hildogjr, 2018-03-20:
Yes.
I am using the (last) nightly version of KiCad. If I add a 0805 capacitor, the 3D Shape file is `Capacitors_SMD.3dshapes/C_0805.wrl`, but the correct had to be `${KISYS3DMOD}/Capacitor_SMD.3dshapes/C_0805_2012Metric.step`.
The same is happening with resistors.
This started when the 3D package started to be distributed as a `kicad-package` on Ubuntu, separated from `kicad-footprint`.
Do I have to force update some package?

jkriege2, 2018-03-20:
seems like you need to update your footprint lib. The SMD-C/R/L fotprints have been renamed to contain the metric sice too!

hildogjr, 2018-03-20:
I `apt purge kicad`, manually delete the `usr\share\kicad\` (with all libraries) and re-install, but appear the same.
There is some procedure that I miss?

jkriege2, 2018-03-20:
simply clone the GitHub repo to your local computer and use that. Then you should have the correct footprint files.

hildogjr, 2018-03-20:
I did, delete the `\usr\share\kicad\library'`, `modules` and `modules\packges3D`. But the error still.
On my "PCB Library Tables", each one is configured a `${KIGITHUB}/*.pretty`, with the variable `KIGITHUB`as `https://github.com/KiCad`. This is a tip that help?

hildogjr, 2018-03-20:
I try update `fp-lib-table` using [this in Git](https://github.com/KiCad/kicad-library/edit/master/template/fp-lib-table.for-pretty) but, this file are using `Capacitors_SMD.pretty` instead `Capacitor_SMD.pretty` and other  issues.

jkriege2, 2018-03-20:
You can simply clone the three required repositories
- optional for you: https://github.com/kicad/kicad-symbols
- https://github.com/kicad/kicad-footprints
- https://github.com/KiCad/kicad-packages3D

to the locations that the environment variables KICAD_SYMBOL_DIR, KISYSMOD and KISYS3DMOD point to. Then, if your project has a fp-lib-table (and if you updated the symbols a sym-lib-table) you should update those with the ones supplied in GIT. Then it should work with a current nightly build.

Did that help?
JAN

hildogjr, 2018-03-20:
Yes, I did this, had have this message when start Pcbnew:

    [3D File Resolver] No such path; ensure the environment var is defined
    /usr/share/kicad/modules/packages3d/Capacitor_SMD.3dshapes/C_0805_2012Metric.wrl

But now, I found the problem. The default defined folder for `${KISYS3DMOD}` in KiCad is

   /usr/share/kicad/modules/packages3d/

And the correct is

    /usr/share/kicad/modules/packages3D

The why:
1) Linux is case sensitive;
2) I don't know why had the final `/`, but cause a problem when use `${KISYS3DMOD}/...`.

closed 2018-03-20 by hildogjr

poeschlr, 2018-03-22:
A bit late but here an explanation why you have had problems:

nightlies (for ubuntu at least) already use the correct files (from the version 5 repos) and should put them into the correct place in your file system. But they do not update the fp-lib-table.

This was a conscious decision. KiCad can not know if you have personal libs in the fp-lib-table or if you deleted libs on purpose from it. It is expected that nightly users know how kicad works and follow the mailing list/bug tracker. (Everything you had problems with had been announced there.)

The final / in the path normally should not make problems. (At least not in ubuntu nightlies)

hildogjr, 2018-03-22:
Yes.
Because this, I deleted the `*-li-table` of my $HOME and start KiCad as the first time. So it ask me to create the default table file. This problem was, maybe, residual from last installations that `apt-get` could not update.
The only "issue" really was that the folder here in the Git is `packages3D` and the default configuration on KiCad is `packages3d\`. So, removing `\` and changing to uppercase `D` worked fine.