KiCad/kicad-footprints #449: Why not use IPC footprint names in v5?

mkschreder, 2018-03-21:
I can not understand why kicad has to go against the guidelines for footprint naming convention that have been well defined for years and name all footprints using semi-arbitrary custom names. 

What I mean is for example the IPC-7351B standard footprint naming convention for SMT parts: http://cxem.net/comp/files/comp149_Footprint-Naming-Convention_-Surface-Mount-Components.pdf

These nming conventions have standard identifiers for virtually all standard parts along with recommendations for naming custom parts. 

So why are we not using them for all new footprints?

mkschreder, 2018-03-21:
For example footprints for qfn would then have following standard convention: https://www.pcb-3d.com/?s=qfn&post_types=models

mkschreder, 2018-03-21:
Here is a very detailed document: https://www.pcb-3d.com/wordpress/wp-content/uploads/ipc-sm-782a_amendment_1_and_2.pdf

I'm going to look into generating all of these automatically. That would be really cool. Imagine just typing an ipc name and getting a footprint without any libraries.

jkriege2, 2018-03-21:
I think there are/were several reasons for choosing the naming scheme:
1. main argument was: our naming scheme is more verbose (e.g. it contains units, it names the numbers that are given (with few exceptions only, many parts are easily human-readable even without looking up, what e.g. "P" stands for ... maybe pitch? or pin?)
1. It's possibly easier for newbies (not electronics experts) to learn/understand
1. It make construction of FPFilters easier (which are an important feature right now in KiCAD)

I'm sure there were more arguments and a discussion can be dug out somewhere ... 

Best,
JAN

mkschreder, 2018-03-22:
Jan, I believe IPC allows you to add extra text after the main footprint name without violating the standard. So I don't see why we can not have both. The standard name is just there to allow scripted creation of the whole footprint from just the name data. 

I think a better solution however would be to make filters work on documentation files that can contain more information. Alternatively extra footprint fields that describe the footprint. I have done a few initial tests on parsing and generating footprints from ipc names and my current solution for resistors basically parses the size out of the ipc name and then generates several footprints based on density (L M or N) requirements. The input is just a list of IPC footprints as CSV.

I think the ideal scenario would be that all standard footprints would be generated from a single csv file that contains ipc names and possibly minimal extra information about the footprint such as stuff that goes into the doc string. Adding new footprints would then be just a matter of adding a single line to the csv file. We could also offer an ipc footprint creation wizard further down the line that would allow one to generate standard footprints with a graphical interface. But I think the most basic first step would be to have support for generating all ipc footprints just from the IPC string.

poeschlr, 2018-03-22:
There will be no further changes possible before the v5 release (not enough time) The discussion about footprint naming was in the summer you could have chimed in back then.

---

We have the same data in the footprint names as IPC also has. (Formatted a bit differently to make it more human readable. Human readability is one of the core ideologies of kicad.)

---

As we (the maintainers) do not have official access to ipc standards we really can not know if any of the footprints are complaint with a particular density level. So even if a contributor states that a particular footprint complies with some IPC standard we can not check if this is true.

For this reason it would be unwise to give the impression that the lib is complaint to IPC. (We try to align our rules to it as far as we can.)

Ratfink, 2018-05-04:
Closing, because the original question was answered by the library admin and there has been no further discussion for over a month.  If we wish to re-consider this topic for the KiCad 6 development cycle, we can open a new issue then.

closed 2018-05-04 by Ratfink