KiCad/kicad-footprints #462: KLC: Silk under SMT part bodies

evanshultz, 2018-03-31:
@poeschlr @jkriege2 @Ratfink @Misca1234 @Shackmeister 
From  https://github.com/KiCad/kicad-footprints/pull/442.

I think KLC needs a bit of clarification about silk going under pads. Perhaps it can be allowed if the package body does not touch the PCB. If the purpose is to prevent silk affecting coplanarity, this does not apply if the part body does not touch the PCB surface where the silk is.

If we still don't want silk under the body, even if the body doesn't touch the PCB, that's OK. But would you also look at the silk pin 1 mark for BGAs and see if there's a standard style we want to adopt? Please take a look at the IPC doc I linked in the PR above.

Ratfink, 2018-03-31:
I agree that it shouldn't be a problem to have silkscreen under a surface mount device where the body doesn't touch the board.  This could also allow silkscreen lines under gull-wing leads in some footprints.  Frankly, I thought this rule was a bit too strict ever since KLC 1.0.

With regards to BGA ball A1 style, I'd prefer the suggestion of removing the silk lines on both sides at the corner, but I wouldn't object to putting a chamfer under the corner if others prefer.

evanshultz, 2018-03-31:
For ease of reference, here is the IPC doc: http://www.ocipcdc.org/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf#page=17 (pages 17, 19, 22).

I like removing both sides in the pin A1 corner too. If the other librarians agree I'll volunteer to make the changes to those footprints.

And KLC should also be updated for both items: silk can be under a SMT body if the body doesn't touch the PCB surface, and also provide an example of our preferred BGA pin 1 silk mark. @poeschlr how do you want to handle KLC updates? Oliver would usually make the updates himself, sometimes after soliciting feedback for proposed wording. I feel there are quite a number of KLC updates discussed here and there but few updates have been made recently. What method do you prefer to make KLC updates?

poeschlr, 2018-04-02:
I am not really sure i understand the proposed silk marker for bgas. could one of you post a screenshot that shows the newly proposed look?

Ratfink, 2018-04-02:
@poeschlr I'm at the wrong computer right now to make an example footprint, but it would be like this image, except both silkscreen lines would be cut back at the top left corner instead of just the one.

[image]

poeschlr, 2018-04-02:
Why would this need a KLC change about allowing silk below the part? Or is this a different unrelated topic?

Ratfink, 2018-04-02:
It's a semi-related topic, because another alternative BGA pin-1 mark on page 17 of [this PDF](http://www.ocipcdc.org/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf#page=17) shows a silkscreen line going under the BGA at the corner.  That shouldn't actually cause any problems because the BGA doesn't sit directly on the board.  Making this KLC change would also allow for more solid silkscreen outlines for e.g. DPAK footprints and the switch footprint in #438.

poeschlr, 2018-04-02:
My main concern is that it will require more work by us maintainers. With the previous rule we simply checked for all smt parts if there is silk below the part. Now we need to look at the datasheet and check if the part is touching the surface of the pcb.

I am reluctant to go along with this unless somebody can show a footprint that would really benefit from it. (A screenshot would be best)

We then need an easy to communicate rule. In detail what does "touching" the surface mean in detail. (what is the minimum distance between the body and the pcb.)

evanshultz, 2018-04-02:
KLC says no silk under SMT parts. So if that was allowed, KLC would need to be changed. But I don't think it's incredibly important to do that, just asking since IPC is showing it in 7351C.

However, my original suggestion was to standardize the BGA pin 1 silk mark by cutting back both corners near pin A1. IPC 7351C does recommended cutting back silk so this is still in line with their recommendations. In the current library there are many styles for silk pin 1 mark on BGAs. Currently, KLC does not show a BGA in section [F5.1](http://kicad-pcb.org/libraries/klc/F5.1/) and if we decide what to do then an example could be added.

fauxpark, 2018-04-03:
@Ratfink
> That shouldn't actually cause any problems because the BGA doesn't sit directly on the board.

If the silk follows the fab chamfer then depending on the spacing between the edge of the package and the balls, it may actually intersect or at least get too close to one:
[image]
(VFBGA-49_5.0x5.0mm_Layout7x7_P0.65mm)

Personally I like the idea of cutting back only the left silk as shown above. It is similar to how the QFN footprints are drawn.