KiCad/kicad-footprints #464: PCB antenna

sdalu, 2018-04-02:
It would be a nice addition if footprints for common PCB antenna could be provided: Bluetooth, Lora, Wifi

Found some Texas Instrument application note for 868MHz/915MHz:
* http://www.ti.com/lit/an/swra416/swra416.pdf
* http://www.ti.com/lit/an/swra227e/swra227e.pdf
* http://www.ti.com/lit/an/swra228c/swra228c.pdf

Ratfink, 2018-04-07:
There already are some 2.4 GHz PCB antennas provided in RF_Antenna.pretty (Texas_SWRA117D_2.4GHz_Left.kicad_mod and Texas_SWRA117D_2.4GHz_Right.kicad_mod).  We would welcome more contributions, of course!

closed 2018-05-19 by sdalu

Harvie, 2019-07-23:
Any news on this?

evanshultz, 2019-07-23:
@Harvie 
There are too many things for librarians to do to take requests, which is why this was closed.

If you or anyone else wants to contribute an antenna footprint that would be great.

Harvie, 2019-07-23:
What about this? [KiCAD_PCB_Antenna_Lib.zip](https://github.com/KiCad/kicad-footprints/files/3423427/KiCAD_PCB_Antenna_Lib.zip)

Source:
https://forum.kicad.info/t/kicad-pcb-antenna-footprints/4765/12
https://drive.google.com/file/d/0B38qWbX1O5CzR2Q4XzljY1NlcTg/view

evanshultz, 2019-07-23:
If you have antenna footprints you'd like to submit, please create a PR so they can be reviewed. If you have a specific question about an antenna footprint or how to implement it in the library, then an issue would be a good place to discuss the question.

sdalu, 2019-07-23:
In fact it was closed, because I gave up on the idea of any contribution to this project. As instead of trying to build on every contributing effort, it is generally expected too much from wanabe contributors for which kicad is an occasional tool and electronic not their main discipline. Guess it should be put as a warning to avoid wasting time for everyone.