KiCad/kicad-footprints #466: Resistor 2512 footprint dimensions

Palmitoxico, 2018-04-04:
I've previously discussed this problem in the kicad-devel mailing list, but it is more appropriate to post it here.

I've verified that the current footprint for the 2512 SMD resistor (imperial) doesn't follow the recommended IPC dimensions.

I'm reading the IPC-SM-782 standard section 8.1 subsection 5.0 and here is a table for comparison:

Dim(mm) | IPC | Kicad
------------ | ------------- | ----------
Z | 7.4 | 7.5
G | 3.8 | 4.8
X | 3.2 | 3.4

The G dimension is too large. I can barely solder the resistors in my board (I should have checked it before sending the gerbers).

Thanks,
Augusto Fraga Giachero.

sethhillbrand, 2018-04-04:
Just a quick note that it looks like more than just the 2512 are off IPC specs.

Palmitoxico, 2018-04-04:
Yes, I just spoted this after trying to solder the resistors by hand. Not a very pleasant surprise : (

I couldn't find where the scripts for generating the footprints are located in this repository. Any hints?

evanshultz, 2018-04-04:
The SMD chip generator scripts are at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/SMD_chip_package_rlc-etc.

poeschlr, 2018-04-04:
The footprints in this repo use the formulas of IPC-7315. (Which is newer then IPC-SM-782)
As far as i can tell the IPC-7315 reduced the pad sizes for all devices.

By the way your measurements are wrong. (or you measured the wrong footprint) The current footprint has G=4.84, Z=7.18, X=3.4

I might have made some mistake for parts where i had the spacing between the contacts given. There i used the maximum spacing instead of using the one where the Root mean square of the different tolerances is used. With the later it would result in G=4.33

sethhillbrand, 2018-04-04:
@poeschlr it looks like the heel/toe/side data were copied from table 3-4 for J-leads instead of table 3-5 larger chips (and used row order rather than labels).

Palmitoxico, 2018-04-04:
@poeschlr I forgot to mention that I measured the handsoldering variant, that's why the Z dimension is higher.

evanshultz, 2018-04-04:
For posterity, IPC 7315 above is really IPC 7351.

Palmitoxico, 2018-04-04:
@evanshultz I was about to post it after spending some time searching for this standard.

By the way, I found the IPC-7351 much more harder to understand than the IPC-SM-782.

evanshultz, 2018-04-04:
@Palmitoxico 
Me too!

poeschlr, 2018-04-04:
Hm i don't know how i got 4.33 for Gmin. Now i always get something about 4.53. I seem to be too tired right now. Will look into it again tomorrow.

And yes i  wrote the wrong standard. I even noticed it while researching something but forgot to fix it.
I did not find the mathematical models behind the old standard. So i can't quite tell if that model was easier. (All i found was a document that had a few dimensions for some common landing patterns in it. No mention of how to derive other landing patterns.)

poeschlr, 2018-04-04:
I think i might have fixed it in the script. https://github.com/pointhi/kicad-footprint-generator/pull/119

I will check it again tomorrow before i create the pull request for the resulting footprints.

poeschlr, 2018-04-05:
The fixed footprints are found in PR https://github.com/KiCad/kicad-footprints/pull/469

evanshultz, 2018-06-07:
As I noticed at https://github.com/KiCad/kicad-footprints/pull/639#issuecomment-395523169, I looked at 0603 and this is what I found (using Level B):
- The equations and processes in our scripts appear to be to be correct
- Zmax = 2.42913mm
- Gmin = 0.785029mm
- Zmax and Gmin should be rounded to 0.01mm. I don't see any guidance about how they should be rounded (floor, ceiling, etc.) but I assume floor for "max" calcs and ceiling for "min" calcs.
- Zmax is obvious at 2.43mm.
- For Gmin I chose 0.79mm the "min" in Gmin means minimum so we shouldn't go below (see above).
- Our pads at Gmin=0.78mm are less than the calculation with 6 significant digits above.
- That then would require pads at +/-0.805mm where ours are +/-0.8mm.

Does anybody find anything wrong with my calcs or rounding understanding?