KiCad/kicad-footprints #497: HandSoldering vs HandSolder suffix

poeschlr, 2018-04-17:
Currently the KLC mentions both these options.

We have 204 hand solder(ing) footprints in total.
114 of them use the HandSoldering suffix, which leaves 90 with the HandSolder suffix.

Which of the two should we use?

---

Some of them do not include the pad size modifier in their name.
Example of the later is the recently merged PR https://github.com/KiCad/kicad-footprints/pull/488

Ratfink, 2018-04-17:
If footprint names are case-sensitive (I assume they are, but KLC doesn't say one way or the other.  Then again, because footprint names are filenames, on Windows they might not be.), we actually have three suffixes in use.  By a count I just did from `master`:

Suffix | Count
-----:|-----
`_HandSolder` | 90
`_HandSoldering` | 83
`_Handsoldering` | 32
**Total** | **205**

We should definitely stick with one suffix for these.  `_HandSolder` does win for brevity, so if we want the path of least resistance, I'd suggest just changing everything (including KLC!) to that.  But I agree with Evan's points above about the ambiguity/subjectivity of the whole thing.  Here's a thought: if we list pad sizes explicitly for all these modified footprints, why would we even need a suffix like `_HandSolder` in the name?

Interestingly, of the 88 of these footprints that list pad sizes, all of which use the `_HandSolder` suffix.  We'd only be adding pad sizes to 117 footprints, which is still a lot, but at least it's better than 205.

stambaughw, 2018-04-17:
What about just "Hand"?  When there were "Reflow" and "Wave" variants of
the SMT resistors and capacitors, we didn't use "ReflowSolder" and
"WaveSolder".  For the sake of consistency, we should pick one style.
My preference is just "Hand" as some of our file names are rather
lengthy.  I could live with "HandSolder" as well.  "HandSoldering" isn't
any more descriptive than "HandSolder".

poeschlr, 2018-04-17:
Adding the pad size is a requirement with the new naming convention. (And yes i know this will be a lot of work.)

---

The HandSolder (or whatever) suffix is there to give the user a reasoning as to why this particular footprint has a differently sized pad to the default footprint. (This suffix clarification is mainly aimed at beginners i think so it should be easily understandable.)

If footprints with the HandSolder suffix really already have the correct name in other regards, keeping this suffix might generate the least amount of work. So my vote would be for this suffix. (To be fair i am a bit biased as a lot of these footprints are generated by my scripts.)