KiCad/kicad-footprints #500: Molex MiniFit Vertical PTH connectors have incorrect pin positions

Scottapotamas, 2018-04-19:
I'm referring to the Molex 5566 vertical connectors. This issue applies to those with, and without, the mechanical mounting pegs.

These connectors are 4.2mm pitch in both x and y as per the datasheet (and the ones in front of me).

[image]

The footprints for the 2x01 through to 2x12 vertical all have the correct 4.2mm pitch in the x-axis, but y-axis pitch is incorrectly set to 5.5mm (same as the RA PTH variants of these connectors.)

[image]

This is pretty visible with the markings on the fab layer, and can be validated by attempting to align a step model against the footprint.

[image]

poeschlr, 2018-04-19:
These are the datasheets for the parts that are compatible to these footprints:

https://www.molex.com/pdm_docs/sd/039281043_sd.pdf
https://www.molex.com/pdm_docs/sd/039289068_sd.pdf

What datasheet is your screenshot from?

poeschlr, 2018-04-19:
I just checked [all 02x02 parts on the official molex website](https://www.molex.com/molex/products/listview.jsp?query=5566&offset=0&npp=20&sType=s&fs=&channel=products&autoNav=0&path=+inmeta%3Apromotable%3Dyes+inmeta%3ACollectionName%3DImpulse+inmeta%3Acategory%3DPCB%252520Headers+inmeta%3Acircuitsloaded%3D4%26requiredfields%3D%28engineeringnumber%3A5566%7Coldpartnumber%3A5566%7Cproductseries%3A5566%29&rangeFlag=true&itemList=) from the 5566 series. Non of them seem to have any other y pitch then 5.5mm. Is it possible you have a knockoff?

Scottapotamas, 2018-04-19:
Its possible families were confused, but when doing the [parametric search on the Molex site ](https://www.molex.com/molex/products/listview.jsp?query=&offset=0&filter=&fs=&npp=20&sType=a&autoNav=&path=+inmeta%3Apromotable%253Dyes+inmeta%3ACollectionName%253DImpulse+inmeta%3Acategory%253DPCB%252520Headers+inmeta%3Aorientation%253DVertical+inmeta%3Anumberofrows%253D2+inmeta%3Acolorresin%253DBlack%2526partialfields%3D%28productname%3AMini-Fit%252520Jr*%29&channel=products&rangeFlag=true&rangeInput=Black&displayName=colorresin&key=minifit_power_connector_solutions) starting from the Mini-fit JR page, then defining vertical and 2-rows it points down to these connectors, I'm not finding any that have a rectangular pattern.

https://www.molex.com/pdm_docs/sd/462071004_sd.pdf is the example which I pulled the screencap off.

The reason I tied this to the 5566 family footprints, is the way Mouser/et al reference the series as 5566, but use manf# such as 39-28-1043 which also have the 4.2x4.2 pattern.

poeschlr, 2018-04-19:
Your datasheet is for the molex minifit jr power **(46207)** series.
Which is not in our library. If you are interested you can adapt the [footprint generator script](https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connector/Connector_Molex/conn_molex_mini-fit_Jr_tht_top_dual-row.py) to generate these as well.

As missing footprints are not an issue worth reporting i will close this. (KiCad does not claim that the library contains every conceivable part.)

closed 2018-04-19 by poeschlr