KiCad/kicad-footprints #525: LGA packages can have different pin layouts. (naming convention)

poeschlr, 2018-04-26:
LGA packages seem to be very flexible with the placement of pins.

I discovered this when i wanted to add my new `LGA-16_3x3mm_P0.5mm` footprint for http://www.st.com/resource/en/datasheet/lis331hh.pdf
There is already a footprint with the same name for http://www.memsic.com/userfiles/files/DataSheets/Magnetic-Sensors-Datasheets/MMC5883MA-RevC.pdf

So my suggestion would be to add a layout specifier similar to what we did with bga footprints.
So something like `_Layout[num pads x rows]x[num pads y rows]`
This would result in the two footprints being named:
- `LGA-16_3x3mm_P0.5mm_Layout4x4`
- `LGA-16_3x3mm_P0.5mm_Layout3x5`

or we make it more clear and use `_Layout[num pads x rows]x[num pads y rows]y`

The pull requests https://github.com/KiCad/kicad-footprints/pull/524 and https://github.com/KiCad/kicad-symbols/pull/546 are affected by this outcome.

After this is done i will also update the old footprints in the lib (plus the symbols that use them)

Ratfink, 2018-04-26:
LGAs seem very non-standard to me, so I wonder if the right approach might be to use manufacturer prefixes.

poeschlr, 2018-04-26:
Well it seems we even found an example for when the same manufacturer uses slightly different footprints. See https://github.com/KiCad/kicad-symbols/pull/545

Ratfink, 2018-04-26:
The problem there is that one package is bigger than the other, so it's not a problem that the current naming can't handle.

poeschlr, 2018-04-27:
Update: i think the layout stuff only makes sense for parts that have pins on all 4 sides. (otherwise the pin count fully specifies the pin layout already)

I just checked the [IPC suggestions](http://ohm.bu.edu/~pbohn/__Engineering_Reference/pcb_layout/pcbmatrix/IPC-7x51%20&%20PCBM%20Land%20Pattern%20Naming%20Convention.pdf)

they suggest to simply include [pin count column]x[pin count row].

---

They also include a pad shape specifier (C = circular pad, R = rectangular, S = square pad) so it seems LGAs at least in theory can have circular and square pads as well.

---

So my updated suggestion would be to use `_Layout[pin x]x[pin y]y` if the pins are arranged similar to a QFN and `_Layout[num colums]x[num rows]` if they are arranged similar to BGA. No suffix for when they are arranged similar to DFN.

For extremely specialized packages we might want to use atomic symbol/footprint connection and naming (both include the part number). At least some pictures on google look like there are LGA packages out there that would be best included that way. (otherwise the naming convention would be extremely complex to include these as well.)

The pin shape can be included as another suffix. I would define rectangular pins as the default and only add a suffix if the leads are circular. (Not sure if we should even include a separate suffix for square leads)

Ratfink, 2018-04-27:
The problem I see with your suggestion is that, for example, the following two footprints could exist:

* `LGA-16_4x4mm_P0.65mm_Layout4x4y`
* `LGA-16_4x4mm_P0.65mm_Layout4x4`

The footprint filter for the latter would match the former.  It would be good to come up with a nomenclature that prevents this.

poeschlr, 2018-04-27:
Argh #$@&%*!
Why is LGA that varied? The qfn like packages don't even have their pins on a single grid most of the time. The array of LGA does not make sense at all for them.

Well enough complaining for today. What do you think about this as a solution?
- `LGA-16_4x4mm_P0.65mm_LayoutBorder4x4y`
- `LGA-16_4x4mm_P0.65mm_LayoutArray4x4` (or grid instead of array)

Or a bit shorter but harder to read (and travis would complain about the footprint filter):
- `LGA-16_4x4mm_P0.65mm_Layout-4x-4y`
- `LGA-16_4x4mm_P0.65mm_Layout4x4`

evanshultz, 2018-04-30:
I like single words and think `Grid` and `Edge` are nice choices.

poeschlr, 2018-05-04:
My LGAs have already been merged. They used the convention LayoutBorder.

So i would in this case codify LayoutBorder and LayoutGrid 
Keeping layout to be similar to BGA, border instead of edge. I chose border because, well i can't really remember anymore.

---

To be sure that the "border decision" was not completely wrong,  i just looked up what google says when searching for border vs edge (i am not a native speaker. So i could have been totally wrong here.)
It seems the edge is the outer outline of somthing whereas the border can be defined as the area around the edge. (As the LGA pads are not really on the edge but only near it i think this might be really the better term.

one definition of border from: https://wikidiff.com/edge/border
> A decorative strip around the edge of something. 

If you guys really object we can still change it as my footprints are scripted anyways.

Ratfink, 2018-05-04:
I don't object to the word "Border" instead of "Edge".  I can justify either:

* The pads are laid out as a border of the footprint.
* The pads are laid out around the edge of the footprint.

I really only favored "Edge" because it was shorter.