KiCad/kicad-footprints #540: Missing footprints for MCU Texas

poeschlr, 2018-04-30:
Footprints missing for https://github.com/KiCad/kicad-symbols/pull/560

 - [ ] 'MSP430F1101AIDGV': http://www.ti.com/lit/ds/symlink/msp430f1101a.pdf
Package_SO:TVSOP-20_4.4x5mm_P0.4mm
 - [ ] 'MSP430G2444IDA38': http://www.ti.com/lit/ds/symlink/msp430g2444.pdf
Package_SO:TSSOP-38_6.1x12.5mm_P0.65mm
 - [ ] 'MSP430F1101AIRGE': http://www.ti.com/lit/ds/symlink/msp430f1101a.pdf
Package_DFN_QFN:VQFN-24-1EP_4x4mm_P0.5mm_EP2.6x2.6mm
 - [ ] 'MSP430G2001IRSA16': http://www.ti.com/lit/ds/symlink/msp430g2001.pdf
Package_DFN_QFN:Texas_S-PVQFN-N16_EP2.7x2.7mm
 - [ ] 'MSP430FR5720IRGE': http://www.ti.com/lit/ds/symlink/msp430fr5720.pdf
Package_DFN_QFN:Texas_S-PVQFN-N24_EP2.2x2.2mm
 - [ ] 'MSP430F1122IRHB': http://www.ti.com/lit/ds/symlink/msp430f1122.pdf
Package_DFN_QFN:Texas_S-PVQFN-N32_EP3.45x3.45mm
 - [ ] 'MSP430F5232IRGZ': http://www.ti.com/lit/ds/symlink/msp430f5232.pdf
Package_DFN_QFN:Texas_S-PVQFN-N48_EP5.15x5.15mm
 - [ ] 'MSP430F5217IRGC': http://www.ti.com/lit/ds/symlink/msp430f5217.pdf
Package_DFN_QFN:Texas_S-PVQFN-N64_EP4.4x4.4mm
 - [ ] 'MSP430F2232IYFF': http://www.ti.com/lit/ds/symlink/msp430f2232.pdf
Package_BGA:Texas_RXBGA-49_3.33x3.488mm_Layout7x7_P0.4mm
 - [ ] 'MSP430F5217IYFF': http://www.ti.com/lit/ds/symlink/msp430f5217.pdf
Package_BGA:Texas_RXBGA-64_3.415x3.535mm_Layout8x8_P0.4mm
 - [ ] 'MSP430F5333IZQW': http://www.ti.com/lit/ds/symlink/msp430f5333.pdf
Package_BGA:Texas_MicroStar_Junior_BGA-113_7.0x7.0mm_Layout12x12_P0.5mm

evanshultz, 2018-07-03:
All of the above will be addressed with https://github.com/KiCad/kicad-footprints/pull/718.

Ratfink, 2018-07-14:
The MCU_Texas_MSP430 library now passes the kicad-library-utils S5.1 check without any errors or warnings, so this issue is resolved.

closed 2018-07-14 by Ratfink