KiCad/kicad-footprints #545: Murata or muRata?

Ratfink, 2018-05-03:
We have a variety of footprints for parts made by [Murata](https://www.murata.com/).  Some include the company's name in the footprint name.  Of these, we have some that capitalize it as `muRata`, and others that capitalize it as `Murata`.  We should reconcile these before the 5.0 freeze begins.

In the company's logo, it's capitalized `muRata`.  However, in ordinary text, they seem to be quite consistent with `Murata`, so I suggest we use the latter.

poeschlr, 2018-05-03:
Acourding to the "about murata" text on the official website the correct writing is "Murata" (muRata is only the logo)

https://www.murata.com/en-us/about?intcid5=com_xxx_xxx_cmn_nv_xxx

Ratfink, 2018-05-04:
The fix for this will involve symbol, footprint, and 3D model changes.  I know for sure it would conflict with https://github.com/KiCad/kicad-packages3D/pull/294, so we should wait for that to be merged before working on this.

Misca1234, 2018-05-06:
The person who undertakes this task should in parallel fix the scripts in 
https://github.com/easyw/kicad-3d-models-in-freecad/pull/161
otherwise that person will get some unneeded workload to guess afterwards which footprint was associated with which 3D model.

Not to mentioned if murata footprints is moved from Converter_DCDC.lib to Regulator_Switching.lib as it should with symbol CRE1S0505S3C

Shackmeister, 2018-06-03:
The 3D model fixes has now been fixed

poeschlr, 2018-08-29:
Has this been done? If not we might want to keep this in mind for new contributions. Existing footprints should really only be renamed if there is a danger of confusion (While we are in the stable release cycle)

JacobEFO, 2019-07-09:
I suppose this issue still needs to be resolved.

I'll be making a PR in a jiffy (both here and on 3d-packages).

antoniovazquezblanco, 2019-07-09:
Thank you @JacobEFO !

antoniovazquezblanco, 2019-07-11:
> Existing footprints should really only be renamed if there is a danger of confusion (While we are in the stable release cycle)

AFAIK there should be no problems if a footprint gets renamed during a stable release although this should not be that necessary.

antoniovazquezblanco, 2019-07-11:
@JacobEFO Is there any symbol that will need footprint or fpfilter updating?

Thanks!

JacobEFO, 2019-07-11:
@antoniovazquezblanco  Can you elaborate this? I don't understand what you mean with "footprint or fpfilter updating".

The changed footprints should be here in this PR.

antoniovazquezblanco, 2019-07-11:
Symbols have fields that point to footprints and footprint filters in order to help the user choose the correct footprint. If you change footprint names, those fields should be updated in the symbols. I did not find any PR updating symbols related to this issue. Should there be one?

Thanks!

evanshultz, 2019-07-11:
This one seems a bit touchy to me to do at this point. Can we at least wait for Rene to return so he can weigh in before merging anything?

antoniovazquezblanco, 2019-07-16:
> This one seems a bit touchy to me to do at this point. Can we at least wait for Rene to return so he can weigh in before merging anything?

No hurry in merging in my side.

@JacobEFO here you can find a list of footprints associations and footprint filters that contain the keyworkd "Murata" in any possible capitalization combination. Some of theese require updating:
```
$ cat *.lib | grep F2 | grep -i murata
F2 "Converter_DCDC:Converter_DCDC_Murata_MGJ3" 0 -450 50 H I C CIN
F2 "Converter_DCDC:Converter_DCDC_Murata_NCS1SxxxxSC_THT" 0 -400 50 H I C CNN
F2 "Converter_DCDC:Converter_DCDC_muRata_OKI-78SR_Vertical" 50 -250 50 H I L CIN
F2 "Converter_DCDC:Converter_DCDC_muRata_OKI-78SR_Horizontal" 50 -250 50 H I L CIN
F2 "Filter:Filter_Murata_BNX025" 0 -200 50 H I C CNN
F2 "Converter_DCDC:Converter_DCDC_muRata_CRE1xxxxxxDC_THT" 0 -350 50 H I C CNN
F2 "Converter_DCDC:Converter_DCDC_muRata_CRE1xxxxxx3C_THT" 0 -400 50 H I C CNN
F2 "Converter_DCDC:Converter_DCDC_muRata_CRE1xxxxxxSC_THT" 0 -400 50 H I C CNN
F2 "Converter_DCDC:Converter_DCDC_muRata_NMAxxxxDC_THT" 0 -350 50 H I C CIN
F2 "Converter_DCDC:Converter_DCDC_muRata_NMAxxxxSC_THT" 0 -350 50 H I C CIN
F2 "Converter_DCDC:Converter_DCDC_muRata_NXE2SxxxxMC_THT" 0 -350 50 H I C CNN
```
```
$ cat *.lib | grep "^ " | grep -i murata
 Converter*DCDC*Murata*MGJ3*
 Converter*DCDC*Murata*NCS1SxxxxSC*
 Converter*DCDC*muRata*OKI*78SR*Vertical*
 Converter*DCDC*muRata*OKI*78SR*Horizontal*
 Filter*Murata*BNX025*
 Converter*DCDC*muRata*CRE1xxxxxxDC*
 Converter*DCDC*muRata*CRE1xxxxxx3C*
 Converter*DCDC*muRata*CRE1xxxxxxSC*
 Converter*DCDC*muRata*NMAxxxxDC*
 Converter*DCDC*muRata*NMAxxxxSC*
 Converter*DCDC*muRata*NXE2SxxxxMC*
```

poeschlr, 2019-09-24:
I would have preferred this to be done after v6 started but as there is already work done i will allow it if everything is done in one go. (including symbol side of things)

cp-aquila, 2020-04-30:
I have something else that we could add here: All parts in `Converter_DCDC`  are also prefixed with `Converter_DCDC`. I suspect that is a leftover from a v4->v5 cleanup. 
I suggest to drop all the prefixes. Perhaps this can be fixed en passant with this issue.

closed 2020-09-28 by myfreescalewebpage