KiCad/kicad-footprints #568: Package_DFN_QFN.pretty inconsistent

cederom, 2018-05-06:
Hey there, looks like DFN_QFN library is inconsistent, detects as INVALID on open, does not want to save because of:

```
IO_ERROR: PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source
'/Users/cederom/(...)/library/kicad/kicad-footprints.git/Package_DFN_QFN.pretty/WQFN-24-1EP_4x4mm_P0.5mm_EP2.7x2.7mm.kicad_mod'
line 31
offset 15
from dsnlexer.cpp : Expecting() : line 369

PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source
'/Users/cederom/(...)/library/kicad/kicad-footprints.git/Package_DFN_QFN.pretty/WQFN-24-1EP_4x4mm_P0.5mm_EP2.7x2.7mm_ThermalVias.kicad_mod'
line 40
offset 15
from dsnlexer.cpp : Expecting() : line 369

PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source
'/Users/cederom/(...)/library/kicad/kicad-footprints.git/Package_DFN_QFN.pretty/QFN-28_4x4mm_P0.5mm.kicad_mod'
line 50
offset 14
from dsnlexer.cpp : Expecting() : line 369
from kicad_plugin.cpp : Load() : line 327
```

I am using KiCAD 4.0.7 on macOS :-)

poeschlr, 2018-05-06:
This lib is for kicad nightly (will become kicad 5)

This is noted in the readme, the repo description on the kicad website and even in the FAQ over at the forum. There is no other place i can think of to add this information.

closed 2018-05-06 by poeschlr

cederom, 2018-05-06:
Maybe nightly but "old stable" `kicad-library` repo is marked as `Archived`.. and still there is no KiCAD5 available.. so it looks this is the "current" footprints library right?

How would exaclty transition from KiCAD 4 -> 5 and libraries would look like?

What library should I use for my current projects made with KiCAD4? Work on a separate `pretty` repo?

poeschlr, 2018-05-06:
Read https://forum.kicad.info/t/how-can-i-install-a-specific-version-of-the-footprint-library/9269/

poeschlr, 2018-05-06:
Or read http://kicad-pcb.org/libraries/download/

cederom, 2018-05-06:
Okay, maybe I did not specify my problem exactly enough, sorry..

I know those documents. I know how to download and use both old and new versions of the libraries. All these information is **user centric**. I want to **contribute** to Symbols and Footprints libraries and those information seems to be missing or outdated. 

I noticed a problem with `Package_DFN_QFN.pretty` inside https://github.com/KiCad/kicad-footprints that prevents me from adding my own components..

The question is how can I contribute components at this point in time where library transitions take place?

I know I cannot use old `kicad-library` repo because it is `archived`. Should I contribute to `kicad-footprints` or separate `Package_DFN_QFN.pretty` repository that is later every some time compiled into `kicad-footprints`?

One simple sentence in the docs on **where** should I **contribute** my components would be really helpful and save some time to newcomers :-)

Ratfink, 2018-05-07:
The footprint libraries are now all in the single https://github.com/KiCad/kicad-footprints/ repository.  All contributions go there.  The old structure with one repository for each footprint library was unmaintainable, so we don't do that anymore.

All contributions should be made with recent development builds of KiCad 5.  That you are using KiCad 4 is most likely the source of the problem you encountered.

There is no `Package_DFN_QFN.pretty` repository, how would contributions go in a repository that doesn't exist? 🙂

I agree that listing the currently used repositories in the [library contribution guide](http://kicad-pcb.org/libraries/contribute/) would be a good idea.  GitHub makes the archived repositories fairly prominent in the [KiCad organization](https://github.com/KiCad) page, so for lack of explicit documentation about this, I find your confusion understandable.

cederom, 2018-05-07:
Thank you @Ratfink :-)

Yea, lots of `*.pretty` repos is not that really pretty :-)

I also could not find a separate repo for `Package_DFN_QFN.pretty` nor `Package_SO.pretty`, as you have noticed, which are most interesting to me at this point thus my further confusion.. the old libraries are not contributeable.. the new does not work with stable release.. thus my questions..

I am working on a project with deadlines. I have decided to use KiCAD instead of new Eagle which I was previously using with a license..

What are my options?

Create my components with stable KiCAD4 and store them in a separate libraries that could be at later point imported into KiCAD5?

Why libraries are not compatible between KiCAD 4 and 5? That seems like a huge danger to compatibility/portability..? Is this really self-compatibility issue or simply a commit bug of some sort?

Is KiCAD5 stable enough to work in a short term timescale without loosing results and not tripling my work?

I cannot see KiCAD in MacPorts.. and right now I really need to focus on my design.. but I would be glad to help you out with KiCAD5 testing in future.. and I would like to contribute components starting today :-)

Ratfink, 2018-05-07:
> Create my components with stable KiCAD4 and store them in a separate libraries that could be at later point imported into KiCAD5?

Yes, that's what you should probably do, if you plan to stick with the stable release of KiCad.

> Why libraries are not compatible between KiCAD 4 and 5?

KiCad 4 libraries work in KiCad 5, but not the other way around.  KiCad 5 has new features, e.g. custom pad shapes, that aren't present in KiCad 4, which is why some footprints from this library don't work in the old version.  There are also non-breaking differences that just annoy us librarians (the new version writes files with a different but compatible line order, so going back and forth would create a lot of noise in Git).

> Is KiCAD5 stable enough to work in a short term timescale without loosing results and not tripling my work?

I wouldn't count on it for anything critical at this point.  I use it for some of my own projects, but it's still getting a lot of changes and will continue to do so at least until the 5.0.0-rc2 release.

cederom, 2018-05-07:
Thank you @Ratfink now all seems clear to me!! Have a nice day!! :-)