KiCad/kicad-footprints #590: Module: Arduino mounting holes too small?

matthijskooijman, 2018-05-17:
Some of the Arduino footprints in the Module library have mounting holes. These holes use a 1.78mm drill size, which seems rather small. For example: https://github.com/KiCad/kicad-footprints/blob/bb3185e1b563b7b3890cd754baaa7d6a8ac32410/Module.pretty/Arduino_UNO_R3_WithMountingHoles.kicad_mod#L96-L99

 Looking at an actual Arduino Uno r3, these have drill holes of around ⌀3mm. Previous (third-party) footprints I've been using also use around 3mm, the [huzza footprint](https://github.com/KiCad/kicad-footprints/blob/master/Module.pretty/Adafruit_HUZZAH_ESP8266_breakout_WithMountingHoles.kicad_mod) has 2.9mm.

@jkriege2, you originally submitted these footprints. Did you intentionally use such small mounting holes?

jkriege2, 2018-05-18:
sounds more like a mistake ... just post a correcting PR!

closed 2018-07-08 by Shackmeister

matthijskooijman, 2018-07-09:
@Ratfink, @Shackmeister, thanks!

PR #735 (2018-07-08): Fix mounting hole size in Arduino Uno footprints
Issue #590 pointed out that it was wrong before.  I looked at the actual Eagle source (from https://store.arduino.cc/arduino-uno-rev3), and found that the correct diameter is 3.2 mm.  This fixes the hole sizes.

Fixes #590.

------------

Thanks for creating a pull request to contribute to the KiCad libraries! To speed up integration of your PR, please check the following items:

- [ ] Provide a URL to a datasheet for the footprint(s) you are contributing
- [ ] An example screenshot image is very helpful 
- [ ] If there are matching symbol or 3D [... cut]