KiCad/kicad-footprints #591: VSSOP-10 footprint

sdalu, 2018-05-19:
There is currently no vssop-10 footprint

This is used for example by the DRV2605 [1] 

[1] http://www.ti.com/lit/ds/symlink/drv2605l.pdf

poeschlr, 2018-05-19:
Why make an issue instead of a pull request?

closed 2018-05-19 by poeschlr

sdalu, 2018-05-19:
Because I'm not good enough at making footprint

poeschlr, 2018-05-19:
And you expect somebody else to do your work?

sdalu, 2018-05-19:
Obviously you don't care about people trying to make contribution at there level of expertise.

poeschlr, 2018-05-19:
It doesn't help us much if people come here and report missing things. If i sit down for half an hour i bet i can list hundreds of missing footprints or symbols.

If we now accept issues for such missing things we get hundreds of open issues within a short amount of time. This will make our life harder then it needs to be.

If you have questions how to make a footprint you are welcome to ask. If it is a library convention specific question ask here if it is a general how to make a footprint question head over to the forum. (Creating footprints/symbols is one of the most important skills an electronic designer must have. Nobody expects your first contribution to be correct at the first try. This is why you would get feedback from us if you make a pull request.)