KiCad/kicad-footprints #595: Minimum annular ring for "big" pads

evanshultz, 2018-05-22:
@poeschlr @jkriege2 @Ratfink @Shackmeister 
We specify the minimum annular ring for vias in KLC, but this doesn't apply for big pads. For example, mounting holes or large pins on components.

See https://github.com/KiCad/kicad-footprints/pull/578 and the part datasheet. For many connectors, only drill hole sizes are given and the annular ring isn't. We currently don't have and standard guidance for contributors.

Using IPC-2221 Level A, and thus being conservative for this type of hole, would require a 0.7mm annular ring width. As this user chose 1mm I think it's fine, but I also think KLC should give a recommendation. Would 1mm be a reasonable minimum annular ring to require for this type of hole?

poeschlr, 2018-05-23:
I took a quick look at IPC-2221 (http://www.sphere.bc.ca/class/downloads/ipc_2221a-pcb%20standards.pdf)

I am a bit confused. Section 9.1.2 (table 9-2) defines the annular rings. The largest one is 0.15mm. (This is the one set in the KLC)
1mm annular ring seems a bit excessive to me. Do you mean 1mm final pad size? If the drill is 0.7mm +2*0.15mm would be 1mm final size. 
1mm annular ring would result in 2.7mm pad size for a 0.7mm drill. So if the 0.7mm drill has a pin pitch smaller than 2.7mm + pad to pad clearance, you can not achieve such a large annular ring. (Annular ring is defined as the smallest copper width anywhere around the pad.)

evanshultz, 2018-05-25:
Table 9-2 applies to vias, not pads where a component lead sits, and that is 0.15mm. Since we can't control pad size on a per-layer basis, and IPC says "All lands and annular rings shall be maximized wherever feasible" on page 73, I believe that means we need to use the external unsupported size on all layers. That is noted in KLC as you mentioned and is the minimum annular ring for any padstack.

But I'm talking about pads for a component lead which also need to consider Table 9-1. After re-reading I think it may mean diameter so using the equation on page 73, and assuming Level A, we get 2 * 0.15mm + 0.4mm = 0.7mm over the drill diameter for the minimum pad diameter.

1mm over the drill size, as I mentioned above, is a bit generous but not overly so. We could use 0.7mm over the drill for a "big" pad and it should be OK per IPC. So... use 0.7mm oversize for pads like this in lieu of any specific pad size given from the part vendor?

And, of course, users can shrink or change the pads in their library or on the board if they like.

poeschlr, 2018-05-27:
Could you make a dimensioned sketch of what you exactly mean? I think that would reduce the chances of misunderstandings.

evanshultz, 2018-05-29:
Sure!

The dimensioning tool in KiCad isn't all that pleasant, but here is what I could quickly make that shows the drill and annular ring dimensions (for some reason I cannot edit the text size of dimensions on the right of the measurement location...):
[image]

The via on the left has a 0.15mm annular ring as the pad diameter is 0.3mm over the drill size. I believe this is all Table 9-2 covers.

The THT hole on the right has a pad diameter 0.7mm over the drill hole. Both Table 9-2 and also Table 9-1 need to be considered. 0.7mm comes from 2 * 0.15mm (minimum annular ring) + 0.4mm (minimum Level A fabrication allowance) as I mentioned above. So going 1mm over the drill size as I originally mentioned would be a bit generous as IPC only says we should go drill diameter + 0.7mm if we want to be conservative (using Level A) and accommodate the largest possible number of PCB suppliers and/or receive the lowest PCB price.

Does that clarify my understanding of the document?

poeschlr, 2018-06-10:
I would not choose level a to be honest. Level b is given as standard and we use the standard (or nominal) dimensions everywhere else.

Remember: too much copper makes soldering harder. It also reduces pad to pad clearances which does reduce the voltage rating of your circuit. It is not simply make copper larger and allow for more alignment tolerances. It is a tradeoff.

---

So for level b we can define the following:
Here L is the maximum lead diameter, a is the drill size given in the pad properties. D is the pad size given in the pad properties. (This would be the absolute minimum. It can be increased all around. There is also the option to make the pads elongated = Oval to get better solder ability for hand soldering. The oval pads would allow to keep the voltage rating high while still making it easier to solder.)

[image]

---

And furthermore ipc-2221a is superseded by ipc-2221b. Sadly i could not find the later standard anywhere. (Ok i could buy it for ~200€ but that is a bit much to invest in a voluntary position.)

Ratfink, 2018-06-10:
Is your calculation 1+2*0.15+0.25=1.5mm an error, or rounding?  Un-rounded, that should be 1.55mm.

poeschlr, 2018-06-11:
@Ratfink 
I guess it is just a typo. (Fixed)

It seems ipc-7351c will do away with the fixed annular ring definition and move over to a definition that depends on the hole diameter.

This document also holds a summary of ipc-7351b recommendations
http://www.ocipcdc.org/archive/What_is_New_in_IPC-7351C_03_11_2015.pdf

evanshultz, 2018-06-18:
I don't have 2221B either.

With mounting holes for large leads which do not have a pad size given, like the connector I linked above, using Level A will give these parts extra strength which is generally what is required. The increase in copper won't be an issue for hand soldering or wave soldering. That is why I suggested Level A and so the arguments for using Level B above don't apply to this stated narrow usage.

In general, Level B is good and I suggest using it by default. But carving out the use of Level A in KLC for mounting pads (especially when no padstack info is provided) seems like a nice allowance to me.

It's not clear to me how 2221B and 7351C (when it's released) will coexist. 7351C is targeted for SMT applications but this section regards THTs. 2221B is quite recent and specifically includes this section for THT pads. I suppose when 7351C is released there will be more information to help guide which standard to use for annular rings.