KiCad/kicad-footprints #596: Is \Sources an accident?

evanshultz, 2018-05-22:
@poeschlr 
The files in https://github.com/KiCad/kicad-footprints/tree/master/Sources seem to be an accidental upload. Am I misunderstanding?

It looks like https://github.com/KiCad/kicad-packages3D-source/tree/master/Package_TO_SOT_THT.3dshapes probably already has the SOD-70 models but I don't see the X2SON one with the rest of the 3D model sources.

poeschlr, 2018-05-23:
No they are on purpose. These are footprint source. (If you open them you will notice that these only have 2d sketches in them. There is no 3d body.) Kicad stepup can export footprints in the latest release.

I can delete them but it will make future maintenance harder. If we get enough of them we can split it into a separate repo but right now i think this would be overkill.

antoniovazquezblanco, 2018-05-23:
I would personally split them and try to gather the script sources for many of the footprints in this repo. There are many footprints currently being generated with python scripts that are hosted outside of the KiCad organization and that could be useful.

poeschlr, 2018-05-24:
I am not aware of any footprint scripts outside of the kicad-footprint-generator repo. Even thought this repo is not part of the kicad organization here on github it is still kind of official.
The same holds true for the 3d generators.

closed 2018-06-04 by evanshultz