KiCad/kicad-footprints #603: Need help for adding Texas Instruments' TPS22993 (WQFN-20)

andrasfuchs, 2018-05-25:
Hi guys,

My [footprint pull request](https://github.com/KiCad/kicad-symbols/pull/588) is getting ready to merge, but we don't have its footprint in the library yet. I would gladly add it, but I haven't created a footprint until now, so I need your assistance, please.

It's the footprint of [TPS22993](http://www.ti.com/lit/ds/symlink/tps22993.pdf).

This is the part of the spec I think we need:
[image]

[image]

It looks like a QFN-20 package to me, so I started to play around with the footprint wizard and selected the QFN script. I could make something similar to what I would need, but I have a few questions.

[image]

First of all, what name should I use when I save the footprint?

I should save it into the Package_DFN_QFN library, right?

What do the "1EP" and "EP1.6x1.6mm" mean in other footprint names?

Should I and if yes, how could I add the NC pad in the four corners of the footprint?

Thanks!

Ratfink, 2018-05-25:
* I'd call this "Texas_S-PWQFN-N20", since this appears to be a manufacturer-specific package.
* You should definitely save this footprint in Package_DFN_QFN.pretty.
* 1EP means the package has one exposed pad, the big thermal pad in the center.  EP1.6x1.6mm gives its dimensions.  This package has no exposed pad, so shouldn't have those fields in its name.
* You should add those corner pads, since they're in TI's example board layout.  I think the right way to handle them would be to give them no pin number.

andrasfuchs, 2018-05-25:
Thanks @Ratfink!

I've created the pull request: https://github.com/KiCad/kicad-footprints/pull/604

closed 2018-05-25 by andrasfuchs