KiCad/kicad-footprints #606: TI-specific footprint names

evanshultz, 2018-05-25:
@poeschlr @jkriege2 @Ratfink 
https://github.com/KiCad/kicad-footprints/pull/604 brought this up again so I think it's worth talking about. We have many TI-specific footprints using different nomenclature. TI often gives various names for the same footprint which can make it hard to know what to call any given footprint.

We should be consistent in how we name the footprints so contributors and users don't have such a burden. With searching in KiCad, we can settle on a nice footprint naming scheme and then include other information in the description and keywords for discovery (plus default footprints, of course).

I confirmed with TI that the 3-letter name (RLW in the link above) is more descriptive and unique than S-PWQFN-N20. So perhaps we should go with that?

Some examples of the confusion:
- `Package_DFN_QFN` has `Texas_RUM0016A` and `Texas_S-PVQFN-N36` and `Texas_VQFN-RGR-20`
- The link above
- https://github.com/KiCad/kicad-symbols/pull/392
- https://github.com/KiCad/kicad-footprints/pull/108
- https://github.com/KiCad/kicad-footprints/pull/122
- https://github.com/KiCad/kicad-footprints/pull/384

evanshultz, 2018-05-25:
Yes, that is what I am saying, according to TI. They tell me the 3 letter nomenclature is the most unique term for a package/footprint.

poeschlr, 2018-08-29:
If we choose to rename footprints now then we need to keep the original one such that we do not break old designs. (Otherwise the lib can not be included in bugfix releases)
Such orphaned footprints could be marked in the description field for later deletion (or we could create a list somewhere)

s-light, 2018-10-19:
any news on this?
i have found some Texas QFN footprints that seems to be not generated by the scripts:
searched in 
[kicad-footprint-generator/..../Package_DFN_QFN/size_definitions/qfn_texas.yaml](https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Packages/Package_DFN_QFN/size_definitions/qfn_texas.yaml)
and
[kicad-footprints/..../Package_DFN_QFN.pretty](https://github.com/KiCad/kicad-footprints/tree/master/Package_DFN_QFN.pretty)
- Texas_S-PVQFN-N40_*
- (and others)

i originally wanted to add a N40 variant with other EP dimensions..
i searched some more and on this journey found the TI -Package Lookup tool:
http://www.ti.com/support-packaging/packaging-tools/find-packages.html

there the 'RTA' Package - also called S-PQFP-N40 (WQFN) i wanted to add linked to
http://www.ti.com/lit/ml/mpqf134a/mpqf134a.pdf
and in this document it is clearly stated that the dimensions of the EP are not specific to this Package but are defined in the Datasheet of the part itself.
**conclusion: we need the EP size in the footprint name**

Question remains: should i add this as a Texas Specific part name or better use the 'normal' name form?
@pointhi can the [QFN-script](https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Packages/Package_DFN_QFN) can generate the D-Shaped Pads as in the Datasheets of these Parts?
[image]
[(screenshot comes from TLC5951 Page 45)](http://www.ti.com/lit/ds/symlink/tlc5951.pdf#page=45&zoom=130,-186,561)

evanshultz, 2018-10-19:
There are many still hand-built. The scripts are relatively new and we've been using them to add new footprints and not replaced existing ones (yet).

Great find on the packaging link from TI! I had not seen this before but surely we'll make good use of it. And it does seem to indicate that the 3-letter code does define uniqueness at TI. Thanks again!

I checked out the datasheets for the RTA package parts (http://www.ti.com/packaging/docs/searchproductbypackage.tsp?orderablePartNumber=&packageDesignator=RTA&pinCount=40&results=results#resultspage) and found a variety of documentation methods:
- Only the general package drawing and no thermal pad size: http://www.ti.com/lit/ds/symlink/adc31jb68.pdf, http://www.ti.com/lit/ds/symlink/dp83848h.pdf, http://www.ti.com/lit/ds/symlink/dp83848j.pdf, http://www.ti.com/lit/ds/symlink/dp83848k.pdf
- 4.15mm square: http://www.ti.com/lit/ds/symlink/amc7823.pdf, http://www.ti.com/lit/ds/symlink/dac7822.pdf, http://www.ti.com/lit/ds/symlink/drv8323.pdf, http://www.ti.com/lit/ds/symlink/tps53647.pdf, http://www.ti.com/lit/ds/symlink/tps53667.pdf
- No package info at all: http://www.ti.com/lit/ds/symlink/ds10br254.pdf, http://www.ti.com/lit/ds/symlink/ds10cp154a.pdf, http://www.ti.com/lit/ds/symlink/ds25br204.pdf, http://www.ti.com/lit/ds/symlink/ds25cp104a.pdf, http://www.ti.com/lit/ds/symlink/lmh1983.pdf
- 4.6mm square: http://www.ti.com/lit/ds/symlink/ds90ub902q-q1.pdf, http://www.ti.com/lit/ds/symlink/ds90uh927q-q1.pdf, http://www.ti.com/lit/ds/symlink/ds92lx1622.pdf, http://www.ti.com/lit/ds/symlink/ds92lx2121.pdf, http://www.ti.com/lit/ds/symlink/lmk00308.pdf, http://www.ti.com/lit/ds/symlink/lmk00338.pdf

The last bullet above shows `RTA0040A` but the 4.15mm one doesn't have a corresponding number. In the bottom right corner of the drawings the 4.15mm one has `4206335-2/F` and the last bullet is `4214989/B`. It appears this is just the drawing number and isn't helpful to us either.

So yes, it does seem like the EP size should be added since that's the most clear solution to this mess.

While TI surely uses generic packages (like from Amkor), at least in some cases, I think going specific is good. AFAIK, using `Texas_RTA_EPXxYmm` or the standard QFN package name but prefixed by `Texas_`. I prefer the former since it includes the 3-letter designation.

No, the script cannot currently generate D-shape pads.