KiCad/kicad-footprints #608: Missing DirectFET footprints

Ratfink, 2018-05-26:
To fix https://github.com/KiCad/kicad-symbols/issues/520 for Transistor_FET.lib, we need to create several DirectFET footprints.  All DirectFET footprints are outlined in [International Rectifier Application Note AN-1035](https://www.infineon.com/dgdl/an-1035.pdf?fileId=5546d462533600a40153559159020f76).  The missing packages are listed below to track the progress.

- [ ] Package_DirectFET:DirectFET_MF
- [ ] Package_DirectFET:DirectFET_L6
- [ ] Package_DirectFET:DirectFET_SA
- [ ] Package_DirectFET:DirectFET_S3C
- [ ] Package_DirectFET:DirectFET_SB
- [ ] Package_DirectFET:DirectFET_MC
- [ ] Package_DirectFET:DirectFET_MD

evanshultz, 2018-05-26:
https://github.com/KiCad/kicad-symbols/pull/297

I intended to make a footprint generator that shares input data with the 3D model generator, but it didn't happen yet.

evanshultz, 2018-09-24:
@Shackmeister 
I know somewhere we talked about the location of the script for the DirectFET footprints merged at https://github.com/KiCad/kicad-packages3D/pull/143. You mentioned perhaps the input parameters (YAML file?) could be shared between the footprints and the 3D models. I'm ready to make the footprints now so I wanted to find this conversation.

Alas, I can't find it no matter how hard I look. I also don't find anything in the 3D source or at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts. Argh! Can you point me to the location of the scripts for these 3D models?

Shackmeister, 2018-09-24:
Hi Evan I made the footprints manually and scripted the 3D models.
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/Directfet_SMD

evanshultz, 2019-09-11:
All footprints have been merged and the PR noted in the first comment.

closed 2019-09-13 by chschlue