KiCad/kicad-footprints #617: Missing SO-DIMM footprints

herostrat, 2018-05-29:
Hello guys,

I wanted to ask if there is a reason why the socket library, and more precisely the SO-DIMMsockets are not planned to migrated to this repo (sockets and te-con is marked as done in the TODO.md)?

evanshultz, 2018-05-29:
It's probably just an oversight.

I don't get any hits for "dimm" or "simm" in the new footprint libraries. Can you point out what footprints didn't get migrated? It would be more helpful if you were clear about exactly what was missing, where it was, and where you looked.

I see some memory socket footprints at https://github.com/KiCad/Sockets.pretty that I don't see in the new libraries.

IC sockets are now at https://github.com/KiCad/kicad-footprints/tree/master/Socket.pretty.

TE footprints which don't fall into another category are at https://github.com/KiCad/kicad-footprints/tree/master/Connector_TE-Connectivity.pretty. There are more TE footprints at https://github.com/KiCad/kicad-footprints/tree/master/Connector_FFC-FPC.pretty, for example.

poeschlr, 2018-05-29:
The footprints in that lib are extremely bad and need extensive rework. But they should still be transferred over. They should probably go into other libs though. (Example the plcc sockets should go into the lcc lib with footprint name prefixed by Socket.)

For reference the original library: https://github.com/KiCad/Sockets.pretty

fauxpark, 2018-05-29:
There is an open PR for one of the SODIMMs at #375 and the PLCC sockets appear to be already migrated over. As for the rest, it was probably a combination of them taking too long/too much effort to fix them up according to KLC in order to migrate them over (when most of the action was happening), and a lot of them could be moved to more specific libraries anyway.

herostrat, 2018-05-30:
Somehow the pull-request https://github.com/KiCad/kicad-footprints/pull/375 did not pop up when I was searching for it.

It is exactly the footprint I was talking about.

It is good to know that it is most likely just an oversight, than I can make pull-requests here in the future for them.

Thanks a lot for the answers!

closed 2018-05-30 by herostrat

btply, 2019-07-28:
Still not showing any SODIMM sockets in a fresh install

evanshultz, 2019-07-29:
@btply 
One only needs to follow the trail to see that https://github.com/KiCad/kicad-footprints/pull/1187 is still open and I made comments about it which have yet to be implemented. I would think it's best to close the issue only after at least one suitable footprint had been merged, though.