KiCad/kicad-footprints #631: Missing SSOP-18 footprint

acolomb, 2018-06-05:
Looking for a matching footprint for a Toshiba 62083A 8-channel DMOS transistor array, I noticed that there is no SSOP-18 footprint in the library.

The SSOP-20 footprint at https://github.com/KiCad/kicad-footprints/blob/master/Package_SO.pretty/SSOP-20_4.4x6.5mm_P0.65mm.kicad_mod comes very close judging by the outline, but has two extra pins.

I kindly ask that whoever generated the fantastic SSOP footprints and matching 3D models, please extend the series by this SSOP-18 variant.

A description of the geometry can be found in Toshiba's data sheet, page 9 top half:
https://toshiba.semicon-storage.com/info/docget.jsp?did=29893

Thank you very much in advance!

poeschlr, 2018-06-05:
The footprints are hand made. (By some contributor. Probably years ago.)

We sadly can not take requests for making footprints (or symbols). 
The only reason to accept such a request would be if the footprint was in one of the old libraries but got removed by mistake. Or at least if a symbol that would use that footprint is already in the library.

---

Creating footprints/symbols is a vital skill for every electronics designer. If you want to contribute both the footprint and symbol for this part you are welcome to do so. Just make sure you follow the library convention: http://kicad-pcb.org/libraries/klc/

closed 2018-06-05 by poeschlr

acolomb, 2018-06-05:
I have already adapted the SSOP-20 footprint for my own use. Just not sure about the 3D model stuff, what would I use to edit that?

Please reopen this issue, I will file a pull request to contribute my footprint file. That will fix it properly.

acolomb, 2018-06-08:
Pending PR review at https://github.com/KiCad/kicad-footprints/pull/632

commit b547d6e6 (2018-07-09): Merge pull request #632 from acolomb/ssop-18

Add 18-pin variant of SSOP used by Toshiba (fixes #631)
commit b0810334 (2018-07-14): Revert "Add 18-pin variant of SSOP used by Toshiba (fixes #631)" (#739)