KiCad/kicad-footprints #654: Use of STEP models by default

FabianInostroza, 2018-06-12:
Why the footprints are still referencing the *.wrl models even if there exist the STEP equivalent and KiCad already supports STEP models?

Also excluding the wrl models may reduce the installer size.

poeschlr, 2018-06-12:
If you set the viewer to step kicad simply changes the file ending of the file path set in the footprint.

A more intuitive solution would have been not to include the file ending at all. But because of backwards compatibility this has been the best solution possible.

---

Wrl files also have their place. While step files are used to communicate with mechanic CAD programs (Examples are FreeCad, Catia, ProE, Soldworks,...). wrl is used to communicate with programs for artsy renderings (Blender,  Rhino, Yafaray. ..) The main benefit of wrl is that it includes detailed material information used for shudders. Step does not allow that. (It only includes one color. The diffuse color.)

---

Tldr: Wrl = nice renderings for the marketing guy, Step = the exact model for the engineer.

evanshultz, 2018-06-19:
This seems closed. Reopen if needed.

closed 2018-06-19 by evanshultz