KiCad/kicad-footprints #658: RFC: Rounded rectangle pads

poeschlr, 2018-06-13:
As IPC suggests to use rounded rectangle pads i looked into what type of footprints are viable to convert.
In future we might also want to codify it in the KLC.

One good example are the two terminal smd footprints (resistor, capacitor, ...) For these it helps with allowing more footprints to have silk screen in addition to the original benefits of having rounded rectangle pads.

Other easily convertible footprints are the scripted smd connector footprints.

I hit a roadblock however with the qfn footprints. Especially with the ones that would require the corner pads to be chamfered. As these chamfered pads are in reality polygon pads it means that they can not have  rounded corners. (at least not with the current implementation of custom pads.) Now the question arises what to do with the remaining pads. Does it make sense to round them of anyways?

Another example where i am unsure are the exposed pads. I would assume they should be rounded as well. (But then again i am unsure as the paste stencil which is the main reason for rounding them is managed by separate pads anyways.)

evanshultz, 2018-06-18:
You must be talking about the unreleased IPC 7351C, right?

I assume the QFN with chamfered pads you're talking about is like `QFN-28_4x4mm_P0.5mm`, right?
[image]

In this case the benefits of rounded corners on all other pads would seem to be worthwhile to me and we should go ahead and do that. It will look a little odd, but get us as close to optimal as we can for the time being.

Has anyone reported this limitation to the devs? I don't see an issue when searching at https://bugs.launchpad.net/kicad. If not, I can report it.

From https://www.pcblibraries.com/Forum/rounded-rectangle-radius-is-different-in-altium_topic1581.html it appears the center thermal pad should have rounded corners. Also see the 3rd post at https://www.pcblibraries.com/Forum/rounded-rectangle-pad-shape_topic1462.html for some semi-official recommendations on radius parameters.

poeschlr, 2018-06-18:
Ok then we agree here. In future we will update the KLC to request rounded rectangle instead of normal rectangle pads.
With the size parameters as suggested by IPC (25% and max radius 0.25mm)

poeschlr, 2018-06-18:
I don't think there is as of yet a bug report for that. So we could request it for version 6.

evanshultz, 2018-06-18:
Submitted at https://bugs.launchpad.net/kicad/+bug/1777516.

closed 2018-08-29 by poeschlr