KiCad/kicad-footprints #660: Current 0402 footprint is wrong!

jcyrax, 2018-06-14:
After https://github.com/KiCad/kicad-footprints/pull/642 new 0402 is obviously wrong with spacing of 0.08mm between pads. I found that there is error in size definition file in https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/SMD_chip_package_rlc-etc/size_definitions/size_default_chip_devices_smaller_0603.yaml terminator_spacing_min: 0.1 is wrong, there needs to be 0.4mm, if you look in linked document. 
[image]
Obviously mistake copying values and T value used instead of S. It looks like its been like this for some time now, just recent changes made it obvious.

poeschlr, 2018-06-14:
fix is part of https://github.com/KiCad/kicad-footprints/pull/653

poeschlr, 2018-06-14:
and this pr https://github.com/KiCad/kicad-footprints/pull/656 fixes it for the v4 backport

evanshultz, 2018-06-16:
@jcyrax 
Both PRs above are merged. Can you please confirm this issue is fixed?

jcyrax, 2018-06-18:
Can confirm, this is fixed now.

closed 2018-06-18 by jcyrax