KiCad/kicad-footprints #661: SOT-363_SC-70-6

Misca1234, 2018-06-16:
Foot print 
Package_TO_SOT_SMD:SOT-363_SC-70-6

pin 1,3,4 and 6 are larger in the data sheet for both

https://assets.nexperia.com/documents/data-sheet/BC846BPN.pdf
https://assets.nexperia.com/documents/data-sheet/BAV756S_BAW56_SER.pdf

https://assets.nexperia.com/documents/outline-drawing/SOT363.pdf

[image]

[image]

Should we remake the foot print

poeschlr, 2018-06-16:
I noticed that nxp uses SC-88 as the name for these packages not SC-70.

---

Using IPC-7351b calculation on the first nxp datasheet i would get the following parameters for the pads:

position y = 0.8375mm, pad size x = 1.275mm,  y = 0.4mm

Misca1234, 2018-06-16:
> I noticed that nxp uses SC-88 as the name for these packages not SC-70.

Yes, but I guess people have reused the foot print because it is so similar, I have not search for which original component the foot print was made for, the foot print does not have a data sheet link

Misca1234, 2018-06-16:
In either case we should add "SOT-363_SC-88-6"

Take 
https://assets.nexperia.com/documents/data-sheet/BAV756S_BAW56_SER.pdf
and BAW56W, it "wants" SOT-363_SC-70-6, which is 3 pins

evanshultz, 2018-06-27:
I would not pick a single vendor's document for our reference footprint. See https://github.com/KiCad/TO_SOT_Packages_SMD.pretty/pull/26.

IPC is the way to go, but unfortunately IPC-SM-782 doesn't have reference dimensions for SOT363 and the superseding 7351 doesn't have package drawings. I found a doc (I think it was in Japanese) that had tons of JEDEC outlines, but naturally I didn't save it and I can't find it now.

Once we have industry-standard reference drawings for a package, I favor splitting up the footprints instead of doing combo ones. After v5 is released and we've settled down, of course.

poeschlr, 2018-08-29:
Even if we finally add specialized footprints we should keep the combo ones for backwards compatibility till v6 development is started.