KiCad/kicad-footprints #668: Missing 3D model for Crystal.pretty/Crystal_SMD_3225-4Pin_3.2x2.5mm.kicad_mod ?

sheffieldnick, 2018-06-18:
This footprint references:
`(model ${KISYS3DMOD}/Crystal.3dshapes/Crystal_SMD_3225-4Pin_3.2x2.5mm.wrl`

but that file doesn't exist in kicad-packages3D (github master).

In 4.x releases I was using ${KISYS3DMOD}/Crystals.3dshapes/Crystal_SMD_SeikoEpson_TSX3225-4pin_3.2x2.5mm.wrl but I can't find an equivalent in the 5.x repository?

Thanks

poeschlr, 2018-06-18:
Some of the old 3d models where of too low of a quality to be included in kicad 5.

The deciding point was if a model has a step model and is scaled and positioned such that the scaling in the footprint can be set to 1 and offset/rotation to 0.

---

The models for seiko epson parts are designed in the old wings way not in the new freecad way. 
So they are scaled wrong and there is no way of getting a valid step model for them. (They need to be recreated in a proper MCAD tool.)

Some of the other crystal models could be easily transferred as they exist as a freecad file. (meaning they simply need to be re exported using kicad-stepup)

poeschlr, 2018-08-29:
Pointer to this issue made in the package 3d repo

closed 2018-08-29 by poeschlr