KiCad/kicad-footprints #678: I think my advice for EP sizes was wrong

poeschlr, 2018-06-20:
As you guys might know, i advised using either IPC standards or at least the maximum tolerance for the ep size.
Most datasheets that give a suggested footprint select the nominal size as the size for the pad on the pcb.
I asked over at the forum for advice: https://forum.kicad.info/t/size-of-exposed-pads/11209

A user there found this [TI document](http://www.ti.com/lit/an/slua271a/slua271a.pdf) that explicitly states that the ep size should be the same size as the "slug" (Which i would interpret as "use the nominal size") Sadly that document does not give a reasoning behind this statement. (The user suggested that it could have to do with improving self alignment.)

---

So i guess my interpretation of the IPC standard in this regard was wrong. (Meaning some of the later added footprints with exposed pads might not be ideal. I doubt they will be so wrong as to merit immediate response but i think in future we can go back to using nominal sizes for the EP.)

poeschlr, 2018-06-26:
Yes i am convinced that IPC-7351 only applies to the lead pads.
The only thing mentioned about exposed pads is that for qfn the clearance between it and the other pads should be at least 0.2mm and that to achieve that it is advisable to reduce the heel fillet.

poeschlr, 2018-06-26:
[Microchip seems to suggest EP sizes all over the place.](http://ww1.microchip.com/downloads/en/PackagingSpec/00000049BQ.pdf) (Sometimes the maximum slug size, but also the minimum and also values in between including nominal.) So i would guess it is not that critical.

poeschlr, 2018-06-26:
Yes i would say so. (There will of course be exceptions if there is a good reason given.)

evanshultz, 2018-06-28:
Just one general question: I have seen bridging between metal elements on a package and traces running under the package. Rivets on snap caps, exposed slugs, formed metal "wings", etc. Is using the nominal value the best choice since we only have an unattributed TI doc saying to do it and other docs giving different guidance?

I got a response from TI and they said basically that this doc is intended to be general and is only pointing out to maintain clearance to other pads. The datasheet supersedes this doc.

I'm fine with using the nominal value, but I just wish we had a more authoritative source of info on this...

poeschlr, 2018-06-28:
Using nominal size is only the advice for exposed pads for parts similar to qfn.

Everything else needs to be examined separately. (And as i hinted above even for qfn there will be exceptions. So if a datasheet suggests a different value and the footprint is specialized enough then this datasheet trumps that. For generic footprints one should then look at suggestions by different manufacturers to get a good conclusion. To be honest really good footprints would also require knowledge of the manufacturing process. But that is something users need to do for them selfes as we can not know how our stuff is used.)

evanshultz, 2018-06-29:
After thinking about this, here are a few pathological questions that might come up. Both of these are driven by thinking about how and why Microchip has such huge ranges for some of their parts. This could happen if there are two different packages (or package vendors) used for some parts so the min and max range are actually roughly the nominal sizes that could be seen on the actual parts but instead of being a tolerance range where any value between min and max can happen, there are only two groupings within that large range.
1. What if nominal slug size isn't given?
-- Use the max size? This would account for the question I pondered above.
-- Calculate nominal from min and max? (Assume min and max values are given.) This would not.
2. What if the max is notably bigger than nominal? (Assume nominal and max are given.) If max is, say, 20% bigger than nominal should the max be used? Otherwise we could have a notably overhanging slug.

evanshultz, 2018-06-29:
In general, when the new footprints with nominal sizes are added, should the older footprints be removed? It shouldn't hurt to keep them, but it will inflate our library a bit with unused footprints. And with your script, it's so easy to make new footprints that re-generating any ones we need later is probably better than having footprints built to no standard rotting in the repo.

poeschlr, 2018-06-29:
I thought i removed them in all of my PRs if i forgot to do it in some of them then it is a mistake