KiCad/kicad-footprints #694: Library for LEM LV 25-P

evanshultz, 2018-06-25:
At https://github.com/KiCad/kicad-footprints/pull/620#issuecomment-399798831 I said:

> One other question: Should we have a new library called `Sensor_Voltage` since `LEM_LV25-P` is voltage transducer and not for current measurement? Not in this PR, but as I was looking at this it seems like the correct way to go.

Isn't that correct?

Datasheet: https://www.lem.com/sites/default/files/products_datasheets/lv_25-p.pdf

poeschlr, 2018-06-25:
Well the first thing that i can read in the datasheet is that this is indeed a voltage transducer. So yes it should be in the lib named Sensor_Voltage. (Adding libs is allowed even after the v5 release.)

evanshultz, 2018-06-26:
Right, but then `Sensor_Voltage` would include only one footprint. OK?

Relatedly, we have this library for symbols and it contains only the matching symbol. Just FYI.

poeschlr, 2018-06-26:
I would guess there will come more in future.

evanshultz, 2018-07-01:
Symbol and footprint PRs made and linked above. Closing this now.

closed 2018-07-01 by evanshultz