KiCad/kicad-footprints #711: 0805 rounded rectangle pad change

stambaughw, 2018-06-28:
The recent change to rounded rectangular pads for the SMT caps and resistors made the gap between the pads on the 0805 footprints less than the 0603 footprint.  I check a bunch of datasheets and other resources and in none of them can I find where the recommended gap between pads of the 0805 footprint is less than the 0603 footprint.  The attached image shows an 0603, 0805, and 1206 resistor (left) and capacitor (right) footprints for comparison purposes.  If this is correct, would someone please attach the document that was used to generate these footprints so I can understand why this occurs.
[image]

poeschlr, 2018-06-28:
I looked into it a bit more. It is even easier to understand than what i wrote above.
For both 0805 and 0603 the terminal to terminal distance is listed in IPC-SM-782. (As written in my previous response i checked these against 20 random resistors and capacitors available from farnell.)

Smin is 0.55 fro 0805 and 0.7 for 0603. If we ignore PCB manufacturing tolerances the pad to pad clearance (Gmin) calculates directly from Smin. (In IPC-7351B the heel fillet is given as 0 meaning Gmin will be the same as Smin.)

If we respect manufacturing tolerances (0.1 fabrication tolerance and 0.05 placement as suggested in IPC-7351B) we get the exact result as the current footprints have.

stambaughw, 2018-06-28:
> I already answered to this over on the mailing list
> <https://lists.launchpad.net/kicad-lib-committers/msg00549.html>.
I guess I should subscribe to the library committers mailing list before
I post a message to it ;)

> TlDr: The reason behind this is the tolerance range of the 0805 is much
> larger then the one for the 0603.
> 
> For reference my answer on the mailing list
> 
>     This is because 0805 has larger tolerance ranges when compared to 0603.
> 
>     For 0603 the tolerances are body length 0.2mm, "lead" length 0.25mm
>     For 0805 they are body length 0.3mm, "lead" length 0.5mm
> 
>     I verified the tolerance ranges by checking 20 random resistors on
>     farnell. For both 0603 the components fall into the tolerance ranges
>     given by this ipc document. (I even found some 0805 parts that would
>     require increasing the tolerance ranges even further.)
> 
>     You can look at the old IPC-SM-782 [1[ standard and check for your
>     self. That one still gives a suggested footprint. (The new IPC-7351B
>     standard does no longer do that. It gives equations how do derive
>     the pad sizes from the part sizes.) The suggested footprints in that
>     old standard use the same pad to pad clearance for both 0603 and
>     0805. As that old standard gives no explanation of how they derived
>     that size i can only speculate that their rounding base was larger.
> 
>     There will however be some minor improvements to some of these
>     footprints. IPC-7351B uses slightly different equations compared to
>     IPC-7351. The pull request that updates this can be found at [2]
> 
>     A very similar question arose over at the forum [3] (The later part
>     of the discussion is about footprints. The first part is a
>     misconception on how kicad works.)
> 
>     [1] http://www.tortai-tech.com/upload/download/2011102023233369053.pdf
>     [2] #689 <https://github.com/KiCad/kicad-footprints/pull/689>
>     [3]
>     https://forum.kicad.info/t/v5-heads-up-devs-dont-explain-here-the-upcoming-changes/11123/22
>     https://forum.kicad.info/t/v5-heads-up-devs-dont-explain-here-the-upcoming-changes/11123/22

Thanks for the info.  I found some manufactures who indeed have poor
enough tolerances that Smin (IPC-SM-782 Figure 2) is 0.55mm.  The worst
one I found was 0.6mm (Stackpole Electronics).  The resistor
manufacturers is use are around 1.0mm (at least the ones I use) so I may
have to create my own SMT footprints.  I don't know if it's wise to have
that much extra pad when reflow soldering.

Sorry about the noise.

Cheers,

Wayne

evanshultz, 2018-06-28:
But it is a good point. This has come up a few times because it seems unintuitive. Would it be best to have a few 0805 options in the official lib: one covering all 0805s and one for parts with a tighter dimensional tolerance?

stambaughw, 2018-06-28:
This is probably true for most of the small SMT resistor and capacitor
footprints.  In my case, I primarily use Panasonic and Vishay resistors
for which Smin is 1.06mm and 1.08mm respectively.  This is almost double
the 0.55mm specified in ipc-sm-782b.  Getting the footprint correct for
small SMT devices like an 0805 resistor is important to prevent
manufacturing defects such as tombstoning and billboarding during
reflow.  My fear is users who are not aware of the potential
manufacturing issues will blindly use these footprints because some ipc
standard said so.  Having pads that are too large can cause just as many
problems as pads that are too small.

poeschlr, 2018-06-29:
I now also checked the 0603. The average over the selected examples is remarkably close to the IPC-SM-785 values.

stambaughw, 2018-06-29:
> I rechecked my research document. I originally wrote it with the
> formulas from IPC-7351 (back then i had no access to the newer IPC-7351B)
> When updating the document to the IPC-7351B formulas i get Smin equal to
> about 0.96mm (average) and 0.88mm worst case.
>
> I now also transferred my document over to google docs. here the link to
> it
> <https://docs.google.com/spreadsheets/d/1BsfQQcO9C6DZCsRaXUlFlo91Tg2WpOkGARC1WS5S8t0/edit?usp=sharing>
>
> I will therefore update the 0805 resistors with these new values. (And
> later recheck the other sizes that i took from the old ipc document.)

Thank you Rene for taking a closer look at this.

Wayne

evanshultz, 2018-06-29:
Minor note: Typo of "Length" twice in the first column of each sheet.

poeschlr, 2018-06-30:
For 0805 i created a pull request to get the footprint nearer to real parts. See: https://github.com/KiCad/kicad-footprints/pull/712

poeschlr, 2018-06-30:
I fixed the typos. If i find time while i am in london i will add a few more parts to my research. Maybe i can find some clear clusters that allow us to make a few more targeted footprints. But until then the best option will be the average that i used right now.

evanshultz, 2018-09-19:
Closing since this is resolved. For reference, here are the current 0603, 0805, and 1206:
[image]

closed 2018-09-19 by evanshultz