KiCad/kicad-footprints #713: Expiration of unanswered PR's

Shackmeister, 2018-06-30:
Hi library team

After going through some of the old PR's, I was wondering if we should set an "expiration" time of PR's where the contributor hasn't answered even after multiple pokes from the librarians. Perhaps something like 3 months.
Thoughts?

Ratfink, 2018-06-30:
What's the harm in leaving old PRs open, and what's the benefit in closing them?

Shackmeister, 2018-06-30:
Harm: clutter the repo.
To me it doesn't make sense to have inactive PR's open as they usually don't ever get resolved anyway.
When looking at the forum, it seems many people see's the huge amount of unsolved PR's as a sign of lack of librarian work and as a proof of how hard is it to contribute.
It might just be me who prefers to have the list of PR's as a proper todo, hence the "thoughts"

Ratfink, 2018-06-30:
Okay, it seems that there are at least two schools of thought on this then:

1. "Open" means "active".
2. "Open" means "could be merged, pending review and possibly further work".

Ultimately it isn't an important distinction, since PRs can be re-opened at any time with one click, and librarians still get notifications about updates to closed PRs.  I could argue for either side of this issue, so I'll save myself the effort and let others who have stronger opinions make the decision.

Shackmeister, 2018-06-30:
As you point out, they can always be opened again. I would never close one due to missing review. But after a review has been made and the contributor doesn't reply after 6 months and no reply what so ever. I don't consider the PR active anymore

evanshultz, 2018-06-30:
I was actually thinking something similar just yesterday as a means to catch anything that we wanted to get in for a release. It's hard to keep track (for me) of what still needs to be done when the number of rotting PRs just goes up and up.

As you said, if a contributor doesn't come back after a long time we will never be able to reach 0 open PRs (not that we need to or that it's even a reasonable target, but still...)

Closing a PR makes it somewhat "disappear" and perhaps lowers the chance of it ever getting resolved, either by the original submitter or someone else. I have a document where I capture abandoned PRs I want to work on in the future when time allows. That's how I deal with it wouldn't close a PR without providing advance warning with a request to finish multiple times following a long period of inactivity.

I'm not quite what would be best either...

antoniovazquezblanco, 2018-07-02:
As a small contributor, having that many open PRs looks like you are overwelmed with the amount of work for the existing librarians. I understand that closing them may be sending a negative message. I also think that the negative impresion can be mitigated by leaving a closing comment saying something like the PR is closed due to inactivity. If you combine this with tagging abandoned PRs for pickup in the future I think everything will look a little bit cleaner.