KiCad/kicad-footprints #741: SO-DIMM

herostrat, 2018-07-10:
Hey guys, 

before I start working on a PR for the connector, extending https://github.com/KiCad/kicad-footprints/pull/375, I have a few questions:

The name: it seems that a number of different sockets have the same dimensions.
TE and Amphenol are the only producers I can find at the usual market places.

Here the naming schema for DDR2 SO DIMM and DDR1 SO DIMM are given for TE:
http://www.te.com/commerce/DocumentDelivery/DDEController?Action=srchrtrv&DocNm=1-1773929-8_Memory_Socket_QRG&DocType=DS&DocLang=EN

So I think we have this groupings for SMT:
 - Standart (Boss, RightAngle - RightSide)
   - 1473005 
   - 1717254
   - 1473149
 - Reverse (Boss, RightAngle - LeftSide)
   - 1473150
   - 1565918
 - Bossless (Bossless, RightAngle - RightSide)
   - 1-1473005

Does anybody have a idea for a sensible naming schema for us?

I think the CtrYd should also not extend over the whole area between the arms.
Two reasons: One can place components under the pcb in the socket and the size of the pcb in the socket is not standardized. Therefore I would like to just outline the socket and not expand over the wole area.
The CtrYd and silk for the arms should be the size as they are in the bobc PR in order to account for the different versions imho.

poeschlr, 2018-08-27:
Sorry for the late answer. (I totally missed this issue)

I doubt we will be able to come up with a naming scheme that will work for all possible parts. We can try what you suggest above and see where it takes us.

Regarding courtyard: I agree that we can not know how large the module is that is inserted into this socket. That part of the area would need to be included in a separate (virtual) footprint that is under the responsibility of the user. Maybe place a text on the comments layer to remind the user that there will be something in this area.

herostrat, 2018-12-11:
I decided to use a non generic naming scheme after reviewing (a lot) of different sockets.
The difference in the different versions is mostly the position of the mounting pads under the wings.

If one wants to use a different version of the available sockets it is easy to modify the submitted one in https://github.com/KiCad/kicad-footprints/pull/1158 so that it fits you need.

closed 2018-12-11 by herostrat