KiCad/kicad-footprints #749: Converter_ACDC: IRM-02 SMD footprint has wrong dimensions

antoniovazquezblanco, 2018-07-16:
http://www.meanwell.com/webapp/product/search.aspx?prod=IRM-02

[image]

Instead of that, the current body is 22.3x37mm

antoniovazquezblanco, 2018-07-16:
Moreover it cannot have the same symbol as the non SMD version due to pin numbering mismatches 🤦‍♂️

antoniovazquezblanco, 2018-07-16:
@Misca1234 Can you handle the update of the 3D generation script and 3D model please? :)

Misca1234, 2018-07-17:
Sure, le me have a look at it
Was a while ago

Misca1234, 2018-07-17:
script push is here
https://github.com/easyw/kicad-3d-models-in-freecad/pull/187

3D model push is here
https://github.com/KiCad/kicad-packages3D/pull/342

antoniovazquezblanco, 2018-07-17:
Thank you very much!

antoniovazquezblanco, 2018-07-25:
Footprint fixed

closed 2018-07-25 by antoniovazquezblanco