KiCad/kicad-footprints #759: Inconsisten footprint naming in C&K switches

antoniovazquezblanco, 2018-07-18:
Examples:
`SW_SPST_CK_KSC6xxJLFx`
`SW_CuK_JS202011AQN_DPDT_Angled`

What should we use for https://www.ckswitches.com/?

antoniovazquezblanco, 2018-09-18:
I know this is low-priority and targets v6 but if we define a naming convention, then we will be able to merge new additions without having to rename them in the future.

Alternatives:
- CK (Used in their URL)
- C&K (Used in their legal texts https://www.ckswitches.com/legal/)
- CuK (Only seen in KiCad)

robkam, 2018-09-18:
It should be more generic than that. Leave out the manufacturer name and model. There are multiple brands and part names sharing the same footprints.

antoniovazquezblanco, 2018-09-18:
Thank you for your input @robkam but unfortunatelly some of the footprints in the repository are manufacturer specific so we should anyways reach a consensus about how we name this manufacturer.

poeschlr, 2018-09-18:
Can the ampersand sign be used in footprint names?

poeschlr, 2019-01-01:
I just tested it. & is ok for kicad. It is however maybe a different story for our footprint download page. (& would need to be escaped in urls)

antoniovazquezblanco, 2019-01-04:
I would avoid using it. Footprint include that characters in filenames and could cause problems in some FS types. Given that the ampersand character is not accepted in G1.1 I would not use it. Just in case.