KiCad/kicad-footprints #764: TO-220 footprint pitch is wrong?

evanshultz, 2018-07-20:
@poeschlr 
At https://github.com/KiCad/kicad-footprints/pull/574 you updated the TO-220 footprint names with staggered pins to the new convention. I have concerns if the naming was done right. Or perhaps I'm confused.

Let's take `TO-220-4_P5.08x2.54mm_StaggerEven_Lead5.84mm_TabDown`:
[image]

`P5.08x2.54mm` is, I believe, the distance from pins 1-3 horizontally and then from pins 1-2 vertically. The first dimension is correct, but the second should then be 3.8mm instead of 2.54mm.

The `Lead5.84mm` is the distance vertically from the bottom of the body to the farthest lead, I think. That is 2.05mm + 3.8mm which is nearly 5.84mm.

Many of the footprints deviate in only one dimension and a few match all 3 of the assertions above, so I think this is correct. I read the conversation linked from the PR above but it's not clear to me exactly what is being proposed so I want to check. http://kicad-pcb.org/libraries/klc/F3.5/ contains some information, but it doesn't say the `P<x>x<y>` section is required with staggered leads and `Lead<distance>mm` isn't shown at all. And the fact that the footprint naming is inconsistent made me want to be sure.

I'm going to add a footprint, and the script was never updated for this new convention, so if someone can confirm the above (updating KLC would be nice!) I can address this.

poeschlr, 2018-07-20:
The lead distance is always from body to pin 1. (No matter if it is the farthest or nearest. See note in the picture of the KLC rule you linked.)

It is possible that the pitch measurement is indeed wrong. (I calculated the measurements by hand when i renamed the parts.)

---

It seems the simplifications made by oliver form my original naming convention did mess that up a bit. (My original intent is documented here: https://github.com/KiCad/kicad-library/issues/1447#issuecomment-323785988)

evanshultz, 2018-07-20:
Thanks for the info! So KLC could use an update and/or clarification?

Fixed footprints and a missing one needed for https://github.com/KiCad/kicad-symbols/pull/765 submitted at https://github.com/KiCad/kicad-footprints/pull/765.

jkriege2, 2018-07-21:
Hmmm ... I'm not very happy with the x-pitch ... because we measure it differently depending on whether the pins are staggered, or not ... which does not make too much sense (I think). Therefore I would propose:
- measure x-pitch always as the distance between the pad-centers of neightboring pin-pairs (on the package) 
- the staggeredEven/Odd as a MODIFIER then tells whether there is a y-pitch between the pins and if so, how they are staggered. If the pins are staggered, add the y-pitch,  otherwise the x-pitch is enough.

Best,
JAN

evanshultz, 2018-07-21:
@poeschlr 
See above. I believe I copied the scheme used in the existing footprints, so please advise. Even your original proposal doesn't cover every field used when you re-named these footprints.

poeschlr, 2018-07-22:
Ok i looked at your PR and i fear i did not follow my own rules when i originally renamed the footprints. It seems i indeed used the pin furthest from the body not pin 1 as the reference. (My bad. So my advice above is totally wrong and you had it right the first time. Sorry about the confusion.)

I do not have any strong feelings either way. 
We would either need to update the KLC. Which we need to do anyways. Or we need to make massive changes to the footprint names. I kind of tend towards the former as this will create fewer problems for us when we want to get the changes into 5.0.1.
(If we keep the naming convention as stable as possible it will be possible to continuously roll out library releases and still call it a stable release cycle.)