KiCad/kicad-footprints #769: QFN with pull back "leads". Naming convention?

poeschlr, 2018-07-22:
At least one qfn footprint is designed for pull back leads (leads that to not come out to the body edge as one would normally expect for such packages)
This means that the footprint does not allow for a toe fillet and is therefore incompatible with normal types of qfn footprints.

The question is how should we indicate that for the users.

The one footprint i found so far is: QFN-44-1EP_9x9mm_P0.65mm_EP7.5x7.5mm with datasheet https://www.parallax.com/sites/default/files/downloads/P8X32A-Propeller-Datasheet-v1.4.0_0.pdf (page 35)

Ratfink, 2018-07-22:
According to [this IPC land pattern naming convention document](http://ohm.bu.edu/~pbohn/__Engineering_Reference/pcb_layout/pcbmatrix/IPC-7x51%20&%20PCBM%20Land%20Pattern%20Naming%20Convention.pdf), IPC calls these "Pull-back Quad Flat No-lead", or PQFN for short.  We could do the same.

It's worth pointing out as well that many (most?) of our LGA footprints are actually in this category.

evanshultz, 2018-07-23:
Hmmm... a quick search online shows that some vendors have varying meanings for "PQFN" that often expose some or all of the leads to the package edge:
- https://www.infineon.com/dgdl/an-1136.pdf?fileId=5546d462533600a401535599f0ef10c7
- https://www.nxp.com/docs/en/application-note/AN2467.pdf
- https://www.fairchildsemi.co.jp/Assets/zSystem/documents/collateral/onlineSeminars/Board-Level-Evaluation-of-Power-Quad-Flat-No-Lead-(PQFN)-Packages-PPT.pdf
- https://www.fairchildsemi.com/technical-articles/Board-Level-Evaluation-of-Power-Quad-Flat-No-Lead-PQFN-Packages.pdf
- https://www.fairchildsemi.com/application-notes/AN/AN-4187.pdf
- http://www.onsemi.com/pub/Collateral/TND6224-D.PDF
- http://www.irf.com/product-info/fact_sheet/mosfets-pqfn.pdf
- https://cdn.macom.com/applicationnotes/S2083.pdf

Some of them are vendor-specific, so they'd still be OK, but some are more general packages which would get confusing. Sigh.

poeschlr, 2018-07-23:
How about we add a suffix _PullBackLeads or something similar? This would allow us to keep the manufacturer prefix (in that case QFN) but still communicate this fact to the user. (This can be in addition to using PQFN)

---

Regarding LGA: Yes i know that they are not really LGAs. But it seems the common usage of LGA has changed to include such packages. (My guess is that they are produced in the same manner as the original LGA packages. This is why the sales guys of packaging companies chose to use LGA as the name for them. That name then makes its way into the user documentation ie datasheet.)

---

I also found another name for QFN/DFN: MLF (micro lead frame) This is for example used in atmel documentations (but they list the alias qfn as well. So we should definitely be ok with using qfn here)

evanshultz, 2018-07-23:
I'm fine with that suffix, but even better if it can be shorter (like `_PullBack`).

MLF is a technology from Amkor. See https://amkor.com/packaging/leadframe/microleadframe/. Vendors choose this package but could call it anything they want.

evanshultz, 2018-09-19:
Seems we decided what to do and have committed some footprints using this terminology already.

closed 2018-09-19 by evanshultz