KiCad/kicad-footprints #772: Do we need a handsolder suffix?

evanshultz, 2018-07-23:
We have chip (and maybe other?) footprints with extra large pads that have a _HandSolder suffix. I question the usefulness.

First, "hand solder" is not universal. It could mean many things. Depending on my skill, vision/magnification, iron tip, etc., what may be hand solderable to one person isn't to another. So just this suffix doesn't necessarily help anyone.

Second, there are other uses for extra large pads. Putting chip parts on the bottom of a PCB and then wave soldering them, for example. Same with SO packages (as long as lead spacing isn't so small that bridging is guaranteed, though you'll probably need solder thieving pads at the trailing side too). So in that case, big pads have nothing to do with hand soldering.

I say that suffix is not used. If the pads are larger than "standard" (IPC, datasheet, etc.) then the pad size is specified in the footprint name. By seeing the `_Pad*` suffix a user will know the pads are enlarged without needing the ambiguous `_Hand` suffix as well.

poeschlr, 2018-07-23:
Handsolder pads will only be enlarged toward the outside (so on the toe side)
Wave soldering will be enlarged on all sides (at least heel and toe)

So the HandSolder suffix gives information. (It tells the user the intent behind the enlarged pads)

poeschlr, 2018-08-29:
The main discussion about the suffix can be found in this older issue

closed 2018-08-29 by poeschlr