KiCad/kicad-footprints #774: License exception in LICENSE.md is inconsistent

StefanBruens, 2018-07-23:
The exception reads as following:

_To the extent that the creation of electronic designs that use ‘Licensed Material’ can be considered to be ‘Adapted Material’, then the copyright holder waives **article 3** of the license with respect to these designs and any generated files which use data provided as part of the ‘Licensed Material’._
 
The CC-BY-SA 4.0 has **no articles**, but **sections** (and subsections). During the license text, references are given in the format "Section 2(a)(4)".

poeschlr, 2018-07-23:
@stambaughw i would guess the cern legal department might be required again.

poeschlr, 2018-07-23:
I am also quite reluctant to change it as i am not sure if we would need the agreement of everyone involved again. So if there is no legal reason to change it i would keep it as is.

StefanBruens, 2018-07-23:
I would claim the exception as is is void, there is no "article 3", so nothing is waived.