KiCad/kicad-footprints #775: Please file the license exception with SPDX.org

StefanBruens, 2018-07-23:
SPDX tracks licenses and relevant exceptions, see https://spdx.org/licenses/exceptions-index.html
Filing additional exceptions with SPDX is not much work, see https://spdx.org/spdx-license-list/request-new-license

herostrat, 2019-01-17:
IANOL, so is this issue an extension to your previous report in https://github.com/KiCad/kicad-footprints/issues/774?

StefanBruens, 2019-01-17:
No, its orthogonal.

herostrat, 2019-01-18:
Than I do not get it.

Why and for what would we want and need to file an exception?
Again IANAL, so please be lenient :)

poeschlr, 2019-01-18:
We have an exception in our license. The exception allows users to share derivative work (their designs) without needing to distribute it under the same license.

The question really is: What benefit does it bring to the kicad project to register our exception at some third party?

StefanBruens, 2019-01-18:
SPDX is not "some" third party, but the de facto registry for licenses. Think of IANA, but licenses, not protocols.

SPDX is used by most major open source projects. When you stick to the "basic" licenses, e.g. GPL or BSD variants, there is nothing you have to do, but the ones which carry some exceptions typically have filed it, e.g. LLVM, GCC, WxWidgest, ..., see https://spdx.org/licenses/exceptions-index.html.

The benefit is, you do not have to read each license again and again, checking _if_ there is an exception, and find out what the exception is.

Distributions carry the SPDX identifier(s) in their meta data, and they check if the distributed source matches the metadata.

herostrat, 2019-01-18:
Ah ok, very interesting.

The problem is that I think that this issue here does not reach the necessary organs of KiCad that are able to decide or change anything.
Afaik most legal stuff gets decided by the KiCad maintainers in combination with the legal department of cern (as communicated in the other issue here).

I think a better approach would be to (also) report this to the project, because the library is a part of it.
The place to do this would be launchpad (https://bugs.launchpad.net/kicad/+filebug)
Although I cannot make any promises as to the speed of feedback or action, the issue would at least be known to the lead (Wayne, who also answered in the other issue) and others that do not usually come around here.
If you are already registered there you can file it yourself, otherwise feel free to ping me.

Sorry if you do not get the feedback one would like.
Afaik no one here on the github-crew had anything to say with the licensing (I could be wrong here) and most are more interested in the project itself and not legal stuff (that for layman as myself seem to be not *that* important or breaking)
It is nontheless very kind and important to have reports that deal in this domain too :)

poeschlr, 2019-01-18:
The license was indeed a suggestion by the cern legal team. We on the library side would have preferred the use of the geda exception in combination with gpl. (As far as i am aware the switch to CC was because cern has more experience with that license.)

poeschlr, 2019-01-18:
And something else. SPDX has a legal team that checks the exception. What happens if the library exception is not accepted?  We really can not change it as there are way too many different contributors that would need to accept a new license. (We are already in a gray area as we have changed the license in the past without getting explicit acceptance from everyone. I really do not wish to repeat this again.)

poeschlr, 2019-01-18:
Just to make it clear: I do not think that the kicad library license is invalid. I further do not believe that it does not fulfill the "free" requirement by debian. (The exception grands more freedom to the users of the software than the normal licenses would.)

What i am worried about is that registering this will create work for us maintainers. And to be honest: I am not interested in investing any time into license stuff unless there is clear evidence that there is a major benefit to the project. I have no problem with somebody taking this on. But the project leaders must give the ok even for that.
This will mean: Only ask on the mailing list if you personally are prepared to take on the full task of handling this. Do not ask that others take on this boring legal stuff while we have > 200 pull requests open that really need to be taken care of.