KiCad/kicad-footprints #776: Number of positions does match part number in Samtec HLE footprints

mikelawrence, 2018-07-23:
For example the "Samtec_HLE-108-02-xxx-DV-BE-A_2x04_P2.54mm_Horizontal.kicad_mod" footprint indicates it is a 2x04 footprint but the HLE-108-02 part is a 2x08. The footprint matches the description of 2x04. So this footprint should have been called "Samtec_HLE-104-02-xxx-DV-BE-A_2x04_P2.54mm_Horizontal.kicad_mod". The XX in HLE-1XX indicates the number of positions in a row.

This issue seems to be present in both "Connector_Samtec_HLE_SMD" and "Connector_Samtec_HLE_THT" libraries.

poeschlr, 2018-07-23:
Scripts for these are found here: https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connector/Connector_Samtec/conn_samtec_hle.py

evanshultz, 2018-08-19:
@mikelawrence 
Thanks for the bug report! Fix is merged now.

PR #783 (2018-08-19): Fix manufacturer part number for Samtec HLE Connectors
The part number includes the number of pins per row. Not the overall
number of pins. This fixes #776

Sadly github does not seem to notice that these are simple renames and assumes all old files are deleted and new ones added.

Script PR: https://github.com/pointhi/kicad-footprint-generator/pull/155
commit d246436c (2018-08-19): Fix manufacturer part number for Samtec HLE Connectors

The part number includes the number of pins per row. Not the overall
number of pins. This fixes #776
closed 2018-08-19 by evanshultz