KiCad/kicad-footprints #793: Naming convention for parts with connected pins

poeschlr, 2018-07-29:
PR #667 adds a strange DFN package with pins 1, 2, 5, 6 connected together to a large pad. And it also has an additional pin between pins 3,4. I have neither a good plan for the footprint name nor for the pin numbering scheme.
The onsemi datasheets given do not specify a working pin numbering scheme.
 - http://www.onsemi.com/pub/Collateral/517AU.PDF
 - http://aosmd.com/pdfs/datasheet/AON1611.pdf

Another datasheet i found about the same package (By Diodes): https://www.diodes.com/assets/Datasheets/DMN30H4D0LFDE.pdf so it really is not a manufacturer specific.

evanshultz, 2018-07-30:
Something similar came up at https://github.com/KiCad/Housings_DFN_QFN.pretty/pull/66. Not sure if that solution is good, but it's an example to consider.

myfreescalewebpage, 2019-01-26:
Also similar issue at https://github.com/KiCad/kicad-footprints/pull/1081 where we decided to use manufacturer package name.

@poeschlr  Issue still required to be opened ?