KiCad/kicad-footprints #816: Missing PinHeader & PinSocket Horizontal SMD footprint

herostrat, 2018-08-09:
I just saw that there are no script generated footprints for them.

See for example:
https://eu.mouser.com/datasheet/2/181/M20-792R-1273521.pdf

evanshultz, 2018-08-09:
There are single-row male headers in the lib already:
[image]

But that's not exactly what you're looking for. We don't take requests here so I'll close this item after a couple days.

Our footprint scripts are at https://github.com/pointhi/kicad-footprint-generator, and we usually get the footprints merged before the script, so if you grab some related script and then work with it you can submit footprints and get help. Being able to write scripts to generate footprints is a powerful tool so it would be great to add your contributions to the repository. There are lots of Python tutorials and information online so start by running an existing script and then figuring out how it works; you'll quickly be able to modify it to fit your needs! :)

herostrat, 2018-08-10:
This issue was not meant to be a feature request.

I wanted to make sure that it is ok to implement it, before I do the work.
It could be a good first starting point to make myself familiar with the footprint scripts.

Would it be ok for you @evanshultz to look over this again https://github.com/KiCad/kicad-footprints/pull/616 and merge it if its ok for you, because I forgot to use branches at this time which means I cannot add additional PRs with an up to date repo

poeschlr, 2018-08-16:
You can still open new pull requests even if your local master branch is tied up somewhere else.

I assume your local repo has this repo (the official kicad repo) added as a separate remote. I call this remote upstream. If you named it differently in your setup simply replace upstream with whatever name you chose.
To get a new branch that follows the Head of the master branch simply use the following commands:
```
git fetch upstream
git checkout upstream/master
git checkout -b new_branch_name
git push --set-upstream origin new_branch_name
```

If you have not yet added the offical repo as a separate remote location run the following command:
`git remote add upstream git@github.com:KiCad/kicad-footprints.git` (or `git remote add upstream https://github.com/KiCad/kicad-footprints.git` if you prefer to use https instead of ssh.)

closed 2019-01-23 by herostrat