KiCad/kicad-footprints #817: Is there really no library for switches?

robkam, 2018-08-10:
Other than button switches I don't see footprint libraries for PCB mount switches, e.g. DIP, rotary or toggle?

evanshultz, 2018-08-10:
In the `Button_Switch` repos I see a few slide, DIP, and piano switches. No rotary or toggle that I noticed.

robkam, 2018-08-11:
Thanks, the name confused me.

robkam, 2018-08-11:
The switch footprints ought to be moved to  e.g. Switch_Keyboard, Switch_Button_THT, Switch_Button_SMD, Switch_DIP_THT, Switch_DIP_SMD, Switch_Misc_THT, Switch_Misc_SMD

robkam, 2018-08-11:
Then just take Buttons out of the names so they're Switch_THT and Switch_SMD. 

BTW I did find rotary switches, in Rotary_Encoder library.

poeschlr, 2018-08-11:
It also has buttons in it in addition to switches. So the library title is more or less to be read as Buttons and Switches. (The description string in the footprint library table makes that clear)

Next time we can do a major lib reorganization is shortly before the v6 release (expected in two years or more) until then we can only add new libs but never delete existing ones. (renaming would include deletion!)