KiCad/kicad-footprints #851: QFN with pullback leads

evanshultz, 2018-08-20:
@poeschlr 
How to generate the TQFN-24 footprint for AS1115 at https://ams.com/documents/20143/36005/AS1115_DS000206_1-00.pdf/3d3e6d35-b184-1329-adf9-2d769eb2404f (page 24)? The current QFN generator doesn't seem to support pullback leads, so should I use the LGA generator?

So far I have the following in `qfn-other.yaml` using the QFN generator, but the script fails (besides not supporting pullback leads) and I haven't looked into it yet. If it's better to start with the LGA generator then I'll just move there instead of pursuing this further.
```
TQFN-24-1EP_4x4mm_Pitch0.5mm_EP52.8x2.8mm_ThermalVias:
  device_type: 'TQFN'
  #manufacturer: 'man'
  #part_number: 'mpn'
  size_source: 'https://ams.com/documents/20143/36005/AS1115_DS000206_1-00.pdf/3d3e6d35-b184-1329-adf9-2d769eb2404f'
  ipc_class: 'qfn_pull_back'
  #ipc_density: 'least' #overwrite global value for this device.
  # custom_name_format:
  #body_size_x_min: 9.9
  body_size_x: 4.0
  #body_size_x_max: 10.1
  #body_size_y_min: 9.9
  body_size_y: 4.0
  #body_size_y_max: 10.1
  lead_width_min: 0.18
  lead_width_max: 0.35
  lead_len_min: 0.3
  lead_len_max: 0.4

  EP_size_x_min: 2.7
  EP_size_x: 2.8
  EP_size_x_max: 2.9
  EP_size_y_min: 2.7
  EP_size_y: 2.8
  EP_size_y_max: 2.9
  # EP_paste_coverage: 0.65
  EP_num_paste_pads: [2, 2]
  #heel_reduction: 0.1 #for relatively large EP pads (increase clearance)

  thermal_vias:
    count: [3, 3]
    drill: 0.3
    # min_annular_ring: 0.15
    paste_via_clearance: 0.1
    # EP_num_paste_pads: [2, 2]
    #paste_between_vias: 1
    #paste_rings_outside: 1
    EP_paste_coverage: 0.65
    #grid: [1, 1]
    # bottom_pad_size:
    # paste_avoid_via: False

  pitch: 0.5
  num_pins_x: 6
  num_pins_y: 6
  #chamfer_edge_pins: 0.25
  #pin_count_grid:
  #pad_length_addition: 0.5
  #suffix: '_Pad{pad_x:.2f}x{pad_y:.2f}mm_HandSolder'
  #include_suffix_in_3dpath: 'False'
```

poeschlr, 2018-08-20:
The problem is with the avoid paste over via stuff. If you set paste_avoid_via to false the footprint is generated. The pull back setting works. you can even include a `lead_to_edge` parameter if the "lead" is not directly on the edge.

The generated footprint will look like this if you set paste_avoid_via to False:
[image]

poeschlr, 2018-08-20:
This would be the parameter set:

``` yaml
TQFN-24-1EP_4x4mm_Pitch0.5mm_EP52.8x2.8mm_ThermalVias:
  device_type: 'TQFN'
  #manufacturer: 'man'
  #part_number: 'mpn'
  size_source: 'https://ams.com/documents/20143/36005/AS1115_DS000206_1-00.pdf/3d3e6d35-b184-1329-adf9-2d769eb2404f'
  ipc_class: 'qfn_pull_back'
  #ipc_density: 'least' #overwrite global value for this device.
  # custom_name_format:
  #body_size_x_min: 9.9
  body_size_x: 4.0
  #body_size_x_max: 10.1
  #body_size_y_min: 9.9
  body_size_y: 4.0
  #body_size_y_max: 10.1
  lead_to_edge: 
    minimum: 0
    maximum: 0.15
  lead_width_min: 0.18
  lead_width_max: 0.35
  lead_len_min: 0.3
  lead_len_max: 0.4

  EP_size_x_min: 2.7
  EP_size_x: 2.8
  EP_size_x_max: 2.9
  EP_size_y_min: 2.7
  EP_size_y: 2.8
  EP_size_y_max: 2.9
  # EP_paste_coverage: 0.65
  EP_num_paste_pads: [2, 2]
  #heel_reduction: 0.1 #for relatively large EP pads (increase clearance)

  thermal_vias:
    count: [3, 3]
    drill: 0.3
    # min_annular_ring: 0.15
    paste_via_clearance: 0.1
    # EP_num_paste_pads: [2, 2]
    #paste_between_vias: 1
    #paste_rings_outside: 1
    EP_paste_coverage: 0.65
    #grid: [1, 1]
    # bottom_pad_size:
    paste_avoid_via: False

  pitch: 0.5
  num_pins_x: 6
  num_pins_y: 6
  #chamfer_edge_pins: 0.25
  #pin_count_grid:
  #pad_length_addition: 0.5
  #suffix: '_Pad{pad_x:.2f}x{pad_y:.2f}mm_HandSolder'
  #include_suffix_in_3dpath: 'False'

```

poeschlr, 2018-08-20:
I should also add that i used the generator as it is in my pull request https://github.com/pointhi/kicad-footprint-generator/pull/149

I think i added some functionality to it and made some bug fixes. So check out my branch if you want to generate your footprint.

poeschlr, 2018-08-20:
I have now implemented support for the way this part is dimensioned in the datasheet. (Commit https://github.com/pointhi/kicad-footprint-generator/pull/149/commits/3dfa62dc5b100055fc1800884677b61a23acb247 includes this feature) The new parameter set would be:

```yaml
TQFN-24-1EP_4x4mm_Pitch0.5mm_EP52.8x2.8mm_ThermalVias:
  device_type: 'TQFN'
  #manufacturer: 'man'
  #part_number: 'mpn'
  size_source: 'https://ams.com/documents/20143/36005/AS1115_DS000206_1-00.pdf/3d3e6d35-b184-1329-adf9-2d769eb2404f'
  ipc_class: 'qfn_pull_back'
  #ipc_density: 'least' #overwrite global value for this device.
  # custom_name_format:
  #body_size_x_min: 9.9
  body_size_x: 4.0
  #body_size_x_max: 10.1
  #body_size_y_min: 9.9
  body_size_y: 4.0
  #body_size_y_max: 10.1
  lead_to_edge:
    minimum: 0
    maximum: 0.15
  lead_width_min: 0.18
  lead_width_max: 0.35
  body_to_inside_lead_edge:
    minimum: 0.3
    maximum: 0.4

  EP_size_x_min: 2.7
  EP_size_x: 2.8
  EP_size_x_max: 2.9
  EP_size_y_min: 2.7
  EP_size_y: 2.8
  EP_size_y_max: 2.9
  # EP_paste_coverage: 0.65
  EP_num_paste_pads: [2, 2]
  #heel_reduction: 0.1 #for relatively large EP pads (increase clearance)

  thermal_vias:
    count: [3, 3]
    drill: 0.3
    # min_annular_ring: 0.15
    paste_via_clearance: 0.1
    # EP_num_paste_pads: [2, 2]
    #paste_between_vias: 1
    #paste_rings_outside: 1
    EP_paste_coverage: 0.65
    #grid: [1, 1]
    # bottom_pad_size:
    paste_avoid_via: False

  pitch: 0.5
  num_pins_x: 6
  num_pins_y: 6
  #chamfer_edge_pins: 0.25
  #pin_count_grid:
  #pad_length_addition: 0.5
  #suffix: '_Pad{pad_x:.2f}x{pad_y:.2f}mm_HandSolder'
  #include_suffix_in_3dpath: 'False'
```

[image]

evanshultz, 2018-08-21:
Thanks! PR submitted at https://github.com/KiCad/kicad-footprints/pull/853.

closed 2018-08-21 by evanshultz