KiCad/kicad-footprints #857: Automatic Footprint Libraries?

sarf2k4, 2018-08-23:
Hi,

I am still new in using kicad, been using kicad v4 for months with slow paced learning speed. Just when I wanted to find a Jumper solder bridge, stumbled across kicad v5. I successfully installed kicad v5, I noticed that the wizard is only for 3d models, not for footprints anymore as for kicad v4 previously. 

I checked the global libraries, there seems to be just the same with previous kicad v4, but browsing to the directory manually using windows explorer, there are additions related/packaged alongside with kicad v5. I am sure that there are more than 5 new footprints/packages included since kicad v4. 

My question is, is there a tool that will append a folder of library/packages automatically without the need of adding every single package over and over again?

evanshultz, 2018-08-23:
Questions about how to use KiCad can be posted at https://forum.kicad.info/, and there is a section for scripts which may be able to help in Pcbnew once the Python API is happy again. This repo is specifically to cover only the official KiCad footprint library.

As I understand it you want to place every footprint (or symbol) in a library onto a PCB (or schematic canvas) automatically instead of dropping them down manually. This is not built-in AFAIK.

closed 2018-08-23 by evanshultz

poeschlr, 2018-08-24:
The wizard is no longer necessary as the github plugin is no longer supported for the official library.

For managing local libs use the library manager.
More details here: https://forum.kicad.info/t/how-can-i-install-a-specific-version-of-the-footprint-library/9269