KiCad/kicad-footprints #864: why arduino footprints is mirror-like in the module.pretty.

shizi599, 2018-08-25:


bobc, 2018-09-30:
It's true, the footprints are mirrored.

An arduino looks like this \:

https://store-cdn.arduino.cc/usa/catalog/product/cache/1/image/520x330/604a3538c15e081937dbfbd20aa60aad/a/0/a000066_featured_4.jpg

The footprint looks like this \:

[image]

It is quite obviously mirrored about the X axis. I guess the idea is that if you are creating a shield, you would flip the arduino footprint so that the connectors are on the bottom of the shield.

The KiCad footprints are weird and no one else does it like this, personally I use my own versions.

jkriege2, 2018-09-30:
When I designed these footprint the decision for this "mirroring" was done consciuously: The idea is to see the Arduino as a component like all other components. Then the component should reside on top of the PCB (that's how we make the KiCAD library and if you want it on the bootom ... you just flip, as with all other components/footprints). Therefore it appears "mirrored", because you would have pinheaders on the PCB that you plug the Arduino into.

If you want to create a Shield, the Arduino is usually on the bottom. In that case you can use the footprint+flip, or youcreate a shield template, which simply contains the connectors on the bottom in the right positions

bobc, 2018-09-30:
I assumed it was deliberate.
> The idea is to see the Arduino as a component like all other components

The problem is literally no one actually does that. People design a shield which has the same orientation as the Arduino. If you flip the footprint, the silkscreen is then on the wrong side.

Was omitting the SPI connector deliberate too?

It's a shame, because the official KiCad Arduino footprints are basically useless. I really don't understand why we have to do it a different way to everyone else, based on one person's whim.

herostrat, 2018-10-01:
If you want to design a shield, why would you not use the arduino templates?

I think the argumentation makes sense:
Footprint is something you want to attach to your PCB. Because the arduino uno comes with female pinheader you have to do it that way

Templates give you the ability to design your own shield which attach to the arduino. 

Those are different usecases.
And I really do not think that they are useless. Even if you would have to mirror them.

herostrat, 2019-01-17:
Is this issue settled?

I think important steps and notes from this things are:
 - Verify the differences and usecases of "footprints" vs "templates", define rules with the maintainers
 - Update the documentation accordingly
 - Recheck if all templates and footprints are correctly designed in such a manner
 - Monitor and push the template functionality

I will create a new issue that tracks these points, as I am currently working on templates myself.

---

@shizi599 do you have anything to add or do you disagree with my previous assessment?
If not, feel welcome to close this issue :)